bperez77/xilinx_axidma #99: Assertion `0 <= siginfo->si_int && siginfo->si_int < axidma_dev.num_channels' failed.

SimonLeeCHN, 2019-05-15:
Hi,
the examples run well on my dma loop design
but when i tried to change the dma channels device-id the example axidma_transfer thow an error

```
AXI DMA File Transfer Info:
Transmit Channel: 4
Receive Channel: 3
Input File Size: 0.00 MiB
Output File Size: 0.00 MiB

axidma_transfer: library/libaxidma.c:193: axidma_callback: Assertion `0 <= siginfo->si_int && siginfo->si_int < axidma_dev.num_channels' failed.
```
i changed the s2mm dma channel to 4,mm2s dma channel to 3
[image]

but if i change the s2mm dma channel to 1,mm2s dma channel to 0,it works well
[image]

did i made a mistake in change the id?

GOOD-Stuff, 2019-05-31:
Hello! 
May be it's because in your axidma_chrdev in dmas set for 0 and 1 channels. 
If you change device-id in channels, you also have to change this IDs in axidma_chrdev.
So, you must set dmas like `= <0x7 0x4 0x7 0x3>.`
For more information, I advice to look on [axidma_of.c](https://github.com/bperez77/xilinx_axidma/blob/master/driver/axidma_of.c)

SimonLeeCHN, 2019-06-04:
Thanks GOOD-Stuff
i was follow the steps in the README.md,so i thought the device-id can be arbitrary
[image]

GOOD-Stuff, 2019-06-04:
Hi SimonLeeCHN!

Did my advice helps to solve your problem? If so, can you please close this issue :)

closed 2019-07-24 by SimonLeeCHN