bperez77/xilinx_axidma #101: memcpy function is slow when copy data from dma output buffer

eleICoto, 2019-06-05:
The code I'm using is like this
[image]
And the result is like this
[image]
it seems like reading data from DMA buffer is much slower than writing data to DMA buffer.
Is there an explanation for this phenomena? Or is there a speedup method

liubowen520, 2019-08-14:
I also encountered this problem. Have you solved this problem？

I want to use fpd-dma instead of memcpy, but fpd-dma can't manipulate the address requested by dma_alloc_coherent.

eleICoto, 2019-08-19:
@liubowen520 
https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18842098/Zynq+UltraScale+MPSoC+Cache+Coherency#ZynqUltraScaleMPSoCCacheCoherency-3.1ARCACHE[3:0]andAWCACHE[3:0]

closed 2019-08-22 by eleICoto

liubowen520, 2019-08-27:
Thank you for your reply.  It is very helpful  for me.  

But I meet problem too, while using dma_coherent. 

vdma failed error 10

eleICoto, 2019-08-27:
@liubowen520 what platform are you using? zynq ultrascale or zynq 7000

liubowen520, 2019-08-27:
I use zynq ultrascale , zcu104.

eleICoto, 2019-08-27:
[image]
this step is the most import one, you have to use method 5.2.2 to enable Broadcasting Inner Shareable for zynq ultrascale

eleICoto, 2019-08-27:
@liubowen520 when does this "vdma failed error 10" error pops out?

liubowen520, 2019-08-27:
Thank you for your reply. I would try this right now.

eleICoto, 2019-08-27:
@liubowen520 Make sure you have read every line of that page and understand what they are talking about

lclwoniu, 2021-01-07:
/ {
	
	axidma_chrdev: axidma_chrdev@0 {
    		compatible = "xlnx,axidma-chrdev";
    		dmas = <&axi_dma_0 0 &axi_dma_0 1>;
    		dma-names = "tx_channel", "rx_channel";
		dma-coherent;
	};
&axi_dma_0 {
	dma-coherent;
	status = "okay";
};
test 

Receive Buffer Size: 32.00 MiB

Using transmit channel 0 and receive channel 1.
	Elapsed Time0: 0.07 s

stone-sjj, 2021-03-19:
> @liubowen520 what platform are you using? zynq ultrascale or zynq 7000

HELLO! my platform is zynq 7000, can I solve the problem(copy data from dma output buffer is slowly) by opening DMA cache function?  how should I do ? thank you very much!

hechengyu-gui, 2022-03-22:
> The code I'm using is like this [image] And the result is like this [image] it seems like reading data from DMA buffer is much slower than writing data to DMA buffer. Is there an explanation for this phenomena? Or is there a speedup method

您好，这个问题解决了吗，我遇到和你一样的问题

lclwoniu, 2022-03-28:
是的 设备树设置一下就可以

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2022年03月22日 19:26 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.******@***.***> |
| 主题 | Re: [bperez77/xilinx_axidma] memcpy function is slow when copy data from dma output buffer (#101) |

The code I'm using is like this And the result is like this it seems like reading data from DMA buffer is much slower than writing data to DMA buffer. Is there an explanation for this phenomena? Or is there a speedup method

您好，这个问题解决了吗，我遇到和你一样的问题

lclwoniu, 2022-03-30:
你搜 petalinux 加速缓冲区读模块

是的 设备树设置一下就可以

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2022年03月22日 19:26 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.******@***.***> |
| 主题 | Re: [bperez77/xilinx_axidma] memcpy function is slow when copy data from dma output buffer (#101) |

The code I'm using is like this And the result is like this it seems like reading data from DMA buffer is much slower than writing data to DMA buffer. Is there an explanation for this phenomena? Or is there a speedup method

您好，这个问题解决了吗，我遇到和你一样的问题

jasonisnotmyrealname, 2023-03-25:
> 你搜 petalinux 加速缓冲区读模块 是的 设备树设置一下就可以
> […](#)

按照上述方法测试过了，还是不行。版本petalinux2021。