bperez77/xilinx_axidma #103: Cross compile error

kimanha, 2019-06-20:
Hello,

I have compile errors with the latest buildboot and xilinx linux kernel version. Does anyone can advice how to fix it.

bpmlab2 ~/xilinx/kiman/test/xilinx_axidma $ make CROSS_COMPILE=/home/bpmlab2/xilinx/linux/buildroot/output/host/usr/bin/arm-buildroot-linux-gnueabihf- ARCH=arm KBUILD_DIR=/home/bpmlab2/xilinx/linux/buildroot/output/build/linux-xilinx-v2019.1 driver
make -C /home/bpmlab2/xilinx/linux/buildroot/output/build/linux-xilinx-v2019.1 M=/home/bpmlab2/xilinx/kiman/test/xilinx_axidma/driver modules
make[1]: Entering directory '/home/bpmlab2/xilinx/linux/buildroot/output/build/linux-xilinx-v2019.1'
  CC [M]  /home/bpmlab2/xilinx/kiman/test/xilinx_axidma/driver/axi_dma.o
/home/bpmlab2/xilinx/kiman/test/xilinx_axidma/driver/axi_dma.c:97:34: error: array type has incomplete element type ‘struct of_device_id’
 static const struct of_device_id axidma_compatible_of_ids[] = {
                                  ^~~~~~~~~~~~~~~~~~~~~~~~
/home/bpmlab2/xilinx/kiman/test/xilinx_axidma/driver/axi_dma.c:98:7: error: field name not in record or union initializer
     { .compatible = "xlnx,axidma-chrdev" },
       ^
/home/bpmlab2/xilinx/kiman/test/xilinx_axidma/driver/axi_dma.c:98:7: note: (near initialization for ‘axidma_compatible_of_ids’)
/home/bpmlab2/xilinx/kiman/test/xilinx_axidma/driver/axi_dma.c:97:34: error: ‘axidma_compatible_of_ids’ defined but not used [-Werror=unused-variable]
 static const struct of_device_id axidma_compatible_of_ids[] = {
                                  ^~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
scripts/Makefile.build:305: recipe for target '/home/bpmlab2/xilinx/kiman/test/xilinx_axidma/driver/axi_dma.o' failed
make[2]: *** [/home/bpmlab2/xilinx/kiman/test/xilinx_axidma/driver/axi_dma.o] Error 1
Makefile:1517: recipe for target '_module_/home/bpmlab2/xilinx/kiman/test/xilinx_axidma/driver' failed
make[1]: *** [_module_/home/bpmlab2/xilinx/kiman/test/xilinx_axidma/driver] Error 2
make[1]: Leaving directory '/home/bpmlab2/xilinx/linux/buildroot/output/build/linux-xilinx-v2019.1'
driver/driver.mk:75: recipe for target 'driver/axidma.ko' failed
make: *** [driver/axidma.ko] Error 2

[image]

shuai132, 2019-06-29:
I'm using v2019.1 too, and got same issue.
just #include <linux/of_address.h> in driver/axi_dma.c will be works!
By the way, you'll need fix axidma_chrdev.c at 277 lines, add true/false as third param. like:
`of_dma_configure(dev->device, NULL, true) ;`

kimanha, 2019-07-05:
Thanks! now it compiled.

bpmlab2 ~/xilinx/kiman/test/xilinx_axidma $ make CROSS_COMPILE=/home/bpmlab2/xilinx/linux/buildroot/output/host/usr/bin/arm-buildroot-linux-gnueabihf- ARCH=arm KBUILD_DIR=/home/bpmlab2/xilinx/linux/buildroot/output/build/linux-xilinx-v2019.1 driver
make -C /home/bpmlab2/xilinx/linux/buildroot/output/build/linux-xilinx-v2019.1 M=/home/bpmlab2/xilinx/kiman/test/xilinx_axidma/driver modules
make[1]: Entering directory '/home/bpmlab2/xilinx/linux/buildroot/output/build/linux-xilinx-v2019.1'
  CC [M]  /home/bpmlab2/xilinx/kiman/test/xilinx_axidma/driver/axidma_chrdev.o
  CC [M]  /home/bpmlab2/xilinx/kiman/test/xilinx_axidma/driver/axidma_dma.o
  CC [M]  /home/bpmlab2/xilinx/kiman/test/xilinx_axidma/driver/axidma_of.o
  LD [M]  /home/bpmlab2/xilinx/kiman/test/xilinx_axidma/driver/axidma.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/bpmlab2/xilinx/kiman/test/xilinx_axidma/driver/axidma.mod.o
  LD [M]  /home/bpmlab2/xilinx/kiman/test/xilinx_axidma/driver/axidma.ko
make[1]: Leaving directory '/home/bpmlab2/xilinx/linux/buildroot/output/build/linux-xilinx-v2019.1'

closed 2019-07-05 by kimanha

vipals, 2019-10-02:
FYI: https://patchwork.ozlabs.org/patch/1169887/