bperez77/xilinx_axidma #104: Data Loss （Continuous sending of real-time data）

chris0306, 2019-07-02:
Hi，Brandon。

Now PL transmit stream to PS, in order to test data integrity，we create data 00 01 02 ... FF。data stream is continuous sending.
On PS side, i tried loop call the function below:

 static int data_recv(axidma_dev_t dev, struct dma_transfer *trans)
{
    int rc;
    // Allocate the recv buffer
    trans->output_buf = axidma_malloc(dev, trans->output_size);
    if (trans->output_buf == NULL) {
        rc = -ENOMEM;
        goto ret;
    }

	// Do receive action
 	rc = axidma_oneway_transfer(dev, trans->output_channel, trans->output_buf, trans->output_size, true);
    if (rc < 0) {
        fprintf(stderr, "DMA read write transaction failed.\n");
        goto free_output_buf;
    } else {
		int i;
		unsigned char* ptr = (unsigned char*)(trans->output_buf);
		for (i = 0; i < trans->output_size; ++i) {
			printf("%02x ", ptr[i]);
			if (i % 16 == 15) {
				printf("\n");
			}
		}
		//printf("====next recv data ====\n\n");
	}

  free_output_buf:
    axidma_free(dev, trans->output_buf, trans->output_size);

  ret:
    return rc;
}
But, I find that the data is discontinuous between the call of data_recv.

Any suggestion？ 
Thanks a lot！

formidable01, 2019-08-20:
It would be helpful to see what the transmit side is doing. The problem could be on that end.

gi4you, 2019-11-10:
@ chris0306  Can you share your PS side test code?
I am trying to the same test (my case ADC stream data ) that used axidma_oneway_transfer() function.
For me, axidma_benchmark.c is working fine for ZCU102 board.

Thanks,

eric950722, 2023-04-23:
@chris0306 Can you share your PS test code, I tried using axidma_oneway_transfer() but keep getting RX channel timeout error.