bperez77/xilinx_axidma #105: Can it be used in multi-DMA

visen-yanghx, 2019-07-23:
Hello everyone：
    if this driver is a single DMA hook, I have succeeded.Can it be used in multi-DMA?
This is my device tree.
		axi_dma_0: dma@80010000 {
			#dma-cells = <1>;
			clock-names = "s_axi_lite_aclk", "m_axi_sg_aclk", "m_axi_mm2s_aclk", "m_axi_s2mm_aclk";
			clocks = <&clk 71>, <&clk 71>, <&clk 71>, <&misc_clk_0>;
			compatible = "xlnx,axi-dma-1.00.a";
			interrupt-names = "mm2s_introut", "s2mm_introut";
			interrupt-parent = <&gic>;
			interrupts = <0 89 4 0 90 4>;
			reg = <0x0 0x80010000 0x0 0x10000>;
			xlnx,addrwidth = <0x20>;
			xlnx,sg-length-width = <0x10>;
			dma-channel@80010000 {
				compatible = "xlnx,axi-dma-mm2s-channel";
				dma-channels = <0x1>;
				interrupts = <0 89 4>;
				xlnx,datawidth = <0x100>;
				xlnx,device-id = <0x0>;
			};
			dma-channel@80010030 {
				compatible = "xlnx,axi-dma-s2mm-channel";
				dma-channels = <0x1>;
				interrupts = <0 90 4>;
				xlnx,datawidth = <0x100>;
				xlnx,device-id = <0x0>;
			};
		};
		axi_dma_1: dma@80020000 {
			#dma-cells = <1>;
			clock-names = "s_axi_lite_aclk", "m_axi_sg_aclk", "m_axi_mm2s_aclk", "m_axi_s2mm_aclk";
			clocks = <&clk 71>, <&clk 71>, <&clk 71>, <&misc_clk_0>;
			compatible = "xlnx,axi-dma-1.00.a";
			interrupt-names = "mm2s_introut", "s2mm_introut";
			interrupt-parent = <&gic>;
			interrupts = <0 91 4 0 92 4>;
			reg = <0x0 0x80020000 0x0 0x10000>;
			xlnx,addrwidth = <0x20>;
			xlnx,sg-length-width = <0x10>;
			dma-channel@80020000 {
				compatible = "xlnx,axi-dma-mm2s-channel";
				dma-channels = <0x1>;
				interrupts = <0 91 4>;
				xlnx,datawidth = <0x100>;
				xlnx,device-id = <0x1>;
			};
			dma-channel@80020030 {
				compatible = "xlnx,axi-dma-s2mm-channel";
				dma-channels = <0x1>;
				interrupts = <0 92 4>;
				xlnx,datawidth = <0x100>;
				xlnx,device-id = <0x1>;
			};
		};
		axi_dma_2: dma@80030000 {
			#dma-cells = <1>;
			clock-names = "s_axi_lite_aclk", "m_axi_sg_aclk", "m_axi_mm2s_aclk", "m_axi_s2mm_aclk";
			clocks = <&clk 71>, <&clk 71>, <&clk 71>, <&misc_clk_0>;
			compatible = "xlnx,axi-dma-1.00.a";
			interrupt-names = "mm2s_introut", "s2mm_introut";
			interrupt-parent = <&gic>;
			interrupts = <0 93 4 0 94 4>;
			reg = <0x0 0x80030000 0x0 0x10000>;
			xlnx,addrwidth = <0x20>;
			xlnx,sg-length-width = <0x10>;
			dma-channel@80030000 {
				compatible = "xlnx,axi-dma-mm2s-channel";
				dma-channels = <0x1>;
				interrupts = <0 93 4>;
				xlnx,datawidth = <0x100>;
				xlnx,device-id = <0x2>;
			};
			dma-channel@80030030 {
				compatible = "xlnx,axi-dma-s2mm-channel";
				dma-channels = <0x1>;
				interrupts = <0 94 4>;
				xlnx,datawidth = <0x100>;
				xlnx,device-id = <0x2>;
			};
		};
		};
	axidma_chrdev@0 {
		compatible = "xlnx,axidma-chrdev";
		dmas = <0x34 0x0 0x34 0x1>;
		dma-names = "axidma0_tx_channel", "axidma0_rx_channel";
		dma-id = <0x0>;
		chrdev-name = "axidma0";
		minor-num = <0x0>;
		num-devices = <0x1>;
	};

	axidma_chrdev@1 {
		compatible = "xlnx,axidma-chrdev";
		dmas = <0x35 0x0 0x35 0x1>;
		dma-names = "axidma1_tx_channel", "axidma1_rx_channel";
		dma-id = <0x1>;
		chrdev-name = "axidma1";
		minor-num = <0x1>;
		num-devices = <0x1>;
	};
[image]

visen-yanghx, 2019-07-23:
[image]

linbaiwpi, 2019-09-09:
as far as I know, you should assign each channel of each dma an unique device-id. please change accordingly and let me know in case of any further problem.

JanVerlinden, 2019-09-13:
I made the configuration with the axi_dma_0 (xlnx,device-id = <0x1>;) and axi_dma_1(xlnx,device-id = <0x0>;), 

From the axidma_benchmark  -help i got this:
        -t <DMA tx channel>:                    The device id of the DMA channel to use for transmitting the data to the PL fabric.
        -r <DMA rx channel>:                    The device id of the DMA channel to use for receiving the the data from the PL fabric.

When I  run this, 
   ./axidma_benchmark -t 0 -r 1
the test will complete successfully.

All other combination will fail. I think those inputs have no effect on device id. But only on selecting the proper tx or rx channel.

Is there another way to select the axi_dma_1?

JanVerlinden, 2019-09-13:
I think it has something to do with this part of the kernel tree:

axidma_chrdev: axidma_chrdev@0 {
    compatible = "xlnx,axidma-chrdev";
    dmas = <&axi_dma_0 0 &axi_dma_0 1>;
    dma-names = "tx_channel", "rx_channel";
};

What should this be?

JanVerlinden, 2019-09-13:
Is there a reason of this restriction?
https://github.com/bperez77/xilinx_axidma/blob/master/driver/axidma_of.c#L241

If i change my kernel tree to:
 axidma_chrdev: axidma_chrdev@0 {
        	compatible = "xlnx,axidma-chrdev";
        	dmas = <&axi_dma_0 0 &axi_dma_0 1 &axi_dma_1 2 &axi_dma_1 3>;
        	dma-names = "tx_channel", "rx_channel" ,"tx_channel1", "rx_channel1";
        };

I got this error message:	
 [   43.443851] axidma: axidma_of.c: axidma_of_parse_dma_nodes: 243: Device tree node axidma_chrdev: Phandle 2 in the 'dmas' property has an invalid channel (argument 0).

JanVerlinden, 2019-09-13:
I have tried updating:
- increased the number of channels
https://github.com/bperez77/xilinx_axidma/blob/master/driver/axidma_of.c#L241
- I have changed the check, so 4 channels could pass:
https://github.com/bperez77/xilinx_axidma/blob/master/driver/axidma_of.c#L130

But now I get this error:
[   44.789393] axidma: axidma_dma.c: axidma_request_channels: 651: Unable to get slave channel 2: tx_channel1

chipgeek, 2019-09-13:
Each DMA has two channels - 0 & 1. You can't give them 2 & 3 as in your device tree. Try this:

 dmas = <&axi_dma_0 0 &axi_dma_0 1 &axi_dma_1 0 &axi_dma_1 1>;

Keep the rest the same.

Also, notice in your device tree, the entry for axi_dma_1: dma@80020000 has 2 channels, both with  xlnx,device-id = <0x1>; You'll need to change one of them to  xlnx,device-id = <0x0>;
 in your system-user.dtsi. Similar to this :

&axi_dma_1 {
	dma-channel@80020000 {
		xilnx,device-id = <0x0>;
	};
};

JanVerlinden, 2019-09-13:
Thanks for the tip. Everything works fine now. 
I added the following items to my device tree:

&axi_dma_0 {
    dma-channel@a0000030 {
        xlnx,device-id = <0x1>;
    };
};

&axi_dma_1 {
    dma-channel@b0000000 {
        xlnx,device-id = <0x2>;
    };
    dma-channel@b0000030 {
        xlnx,device-id = <0x3>;
    };
};

And I was able to run the command for the second dma:
sudo ./axidma_benchmark  -t 2 -r 3

chipgeek, 2019-09-13:
Excellent.

visen-yanghx, 2019-09-23:
Excellent.