bperez77/xilinx_axidma #107: DMA transmit transaction hangs

formidable01, 2019-08-14:
We are experiencing an odd behavior that we are having trouble debugging. We are doing one way transfers on the transmit and receive side. The transmit side has wait = false and the receive has wait = true.  After running for a while the transmit side hangs (the one way transfer call does not return). We can then look at the DMA register space and all the flags indicate that the transfer completed but the receive side times out having not received any data.

We are having difficulty isolating the issue but it appears to be some sort of lockup perhaps owing to a race condition maybe in the driver code itself. If you have any ideas of where to look or what to monitor that would be extremely helpful.

SmartElec, 2020-07-31:
same with you

xiayanqian, 2022-11-10:
Hello, I think we share the same problem,  have you sovled it?

shyley, 2024-01-28:
hello,Maybe I Meet the same problem,h ave you solved it?