bperez77/xilinx_axidma #109: Get actual receive transaction length [Question]

antts, 2019-08-30:
The Xilinx DMA Core may finish a transaction earlier than expected/programmed due to a TLAST on the input slave AXI Stream. 

In this case, how can we know the actual transaction length in user-space when the callback is called?

I have tried setting the receive size larger than the actually performed transaction and it indeed works, but I do not know what the real packet length was. It seems that the DMA Core exports this in the S2MM_STATUS register [Xilinx forums](https://forums.xilinx.com/t5/Embedded-Development-Tools/how-to-determine-the-transfer-length-of-AXI-DMA/td-p/715887). 

Is this available in the API somewhere? (I may just be missing it..) 
If not can anyone suggest a workaround to get this info after a transaction?

Thanks!

GOOD-Stuff, 2019-09-02:
Hello!
You should to look at this issue #60 and #41 
tl;dr: API doesn't provide any information about received transaction length because Xilinx DMA framework doesn't process it. 
You may try to use mmap and get access (via offset) to descriptor's registers and calculate transaction's length, but you may not have time to get it before Xilinx AXI DMA driver process transaction.

antts, 2019-09-03:
According to those two issues this seems non-trivial to do through this driver due to non support from the Xilinx driver. 
I might try to read S2MM_STATUS with mmap unto /dev/mem, thank you for your help!