bperez77/xilinx_axidma #112: DMA mask not set

cristianfatu, 2019-10-25:
The following error is printed in dmesg when running a DMA transaction for the first time on Linux 4.19 on Zynq board.
```
axidma axidma: DMA mask not set
```

The data captured on DMA on the first round of transactions looks weird, but on the second run, after the DMA mask is applied by the underlying linux drivers, it is fine.

I applied the following patch to solve the error.

```
From 827484928a72f7bddb7827f24caba97157ba1480 Mon Sep 17 00:00:00 2001
From: Demon000 <demonsingur@gmail.com>
Date: Fri, 25 Oct 2019 14:42:50 +0300
Subject: [PATCH] xilinx-axidma: configure dma on platform device dev

---
 .../meta-user/recipes-modules/xilinx-axidma/files/axidma_chrdev.c       | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/project-spec/meta-user/recipes-modules/xilinx-axidma/files/axidma_chrdev.c b/project-spec/meta-user/recipes-modules/xilinx-axidma/files/axidma_chrdev.c
index e63c0d1..f99639e 100644
--- a/project-spec/meta-user/recipes-modules/xilinx-axidma/files/axidma_chrdev.c
+++ b/project-spec/meta-user/recipes-modules/xilinx-axidma/files/axidma_chrdev.c
@@ -274,7 +274,7 @@ static int axidma_mmap(struct file *file, struct vm_area_struct *vma)
     dma_alloc->user_addr = (void *)vma->vm_start;
 
     // Configure the DMA device
-    of_dma_configure(dev->device, NULL, true);
+    of_dma_configure(&dev->pdev->dev, NULL, true);
 
     // Allocate the requested region a contiguous and uncached for DMA
     dma_alloc->kern_addr = dma_alloc_coherent(&dev->pdev->dev, dma_alloc->size,
-- 
2.7.4
```

shyley, 2019-11-19:
I have the same problem ,too. please can i ask how you solved this problem?

46cv8, 2021-01-03:
For anyone else running into issues like this there appear to be a number of other fixes and changes available as patches at https://github.com/Digilent/Eclypse-Z7-OS/tree/zmod_dac/master/project-spec/meta-user/recipes-kernel/linux/linux-xlnx
Not sure which ones are worthy of pull requests or not as I'm new to xilinx_axidma.