bperez77/xilinx_axidma #116: recv channel error

whnwhn, 2019-12-22:
when i'm running the examples:

./axidma_benchmark -t 0 -r 1 -b 1500 -s 1500-n 300
AXI DMA Benchmark Parameters:
Transmit Buffer Size: 0.00 MiB
Receivxilinx-vdma 40400000.dma: Channel ef3b9010 has errors 40, cdr 0 tdr 0
e Buffer Size: 0.00 MiB
Number of DMA Transfers: 1000 transfers

Using transmit channel 0 and receive channel 1.
axidma: axidma_dma.c: axidma_start_transfer: 301: DMA receive transaction timed out.
Failed to perform the AXI DMA read-write transfer: Timer expired

dantepayne, 2023-12-25:
hey guys,I meet the same problem,do you fix it?

whnwhn, 2023-12-27:
Sorry dantepayne, the problem still dont get solved.

---- Replied Message ----
| From | ***@***.***> |
| Date | 12/25/2023 15:24 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [bperez77/xilinx_axidma] recv channel error (#116) |

hey guys,I meet the same problem,do you fix it?

dantepayne, 2023-12-27:
thranks for your reply

dantepayne, 2023-12-27:
Guess What?It can be work on Zu5,but cannot work on Zynq7000.[For Linux 5.4.0->Suggestion](https://github.com/bperez77/xilinx_axidma/pull/139/files).

whnwhn, 2023-12-27:
Thank you for information!

---- Replied Message ----
| From | ***@***.***> |
| Date | 12/27/2023 11:23 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [bperez77/xilinx_axidma] recv channel error (#116) |

Guess What?It can be work on Zu5,but cannot work on Zynq7000.For Linux 5.4.0->Suggestion.

jimmyxt, 2024-01-08:
Same problem, ZYNQ7000@Petalinux2020.2.

But I use Vivado 2020.1. When update vivado project to 2020.2, it works fine.