bperez77/xilinx_axidma #122: can axidma_display_image work on vdma ?

zskdan, 2020-06-13:
I'm trying to use axidma_display_image to display an image (bitmap image 640 x 480 x 32) on the DISPLAY attached on HDMI to my ARTY-Z7-20. The FPGA base design is the one in https://github.com/Digilent/Arty-Z7-20-base-linux.
I've just replaced AXIDMA_DMA with AXIDMA_VDMA references as in this patch
[0001-display_image-use-vdma.txt](https://github.com/bperez77/xilinx_axidma/files/4773914/0001-display_image-use-vdma.txt)

But It seems that the 2nd `copy_from_user()` [here](https://github.com/bperez77/xilinx_axidma/blob/42ed91e83bc4da1e29149b2be0c6a6b8f4549222/driver/axidma_chrdev.c#L474) is making a kernel oops:

```
Unhandled fault: page domain fault (0x01b) at 0xbe880b58
pgd = dc26c000
[be880b58] *pgd=1d8e6831, *pte=0107b75f, *ppte=0107bc7f
Internal error: Oops - BUG: 1b [#1] PREEMPT SMP ARM
Modules linked in: xilinx_axidma(O) uio_pdrv_genirq
CPU: 1 PID: 1241 Comm: axidma_display_ Tainted: G           O    4.9.0-xilinx-v2017.4 #13
Hardware name: Xilinx Zynq Platform
task: ddb93200 task.stack: dda3a000
PC is at axidma_ioctl+0x7c8/0xb68 [xilinx_axidma]
LR is at axidma_ioctl+0xa88/0xb68 [xilinx_axidma]
pc : [<bf004cc8>]    lr : [<bf004f88>]    psr: a0070013
sp : dda3beb0  ip : 00000007  fp : be880c6c
r10: 00000000  r9 : dda3a000  r8 : dda1b440
r7 : dda3a000  r6 : 00000004  r5 : c0c44c00  r4 : be880b50
r3 : dda3beb0  r2 : 00000000  r1 : 00000007  r0 : 00000007
Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
Control: 18c5387d  Table: 1c26c04a  DAC: 00000051
Process axidma_display_ (pid: 1241, stack limit = 0xdda3a210)
Stack: (0xdda3beb0 to 0xdda3c000)
bea0:                                     00000000 00000000 ddba1808 c01a7820
bec0: dc2bf240 00000000 00000001 dda1b440 000001e0 00000280 00000003 de033280
bee0: 00000046 00000000 dc2b1180 dda3bf88 000e1000 dda3bf88 be880b50 dc13ede8
bf00: dc2b1000 80185708 be880b50 c01dcc84 be880b50 c01dce28 000e1000 00000000
bf20: 00000000 00000000 dda3bf1c 00000000 dc2b1180 00000000 00000000 00000000
bf40: 00000000 dc2b1188 00000000 000e1000 ddba1808 00000001 dda3a000 00000000
bf60: be880c6c c01cdbbc 00000000 00000004 dc2b1000 dc2b1000 80185708 be880b50
bf80: dda3a000 00000000 be880c6c c01dd670 00011908 00000000 000109a4 00000036
bfa0: c0106e24 c0106c60 00011908 00000000 00000004 80185708 be880b50 be880b50
bfc0: 00011908 00000000 000109a4 00000036 00000000 00000000 b6f8e000 be880c6c
bfe0: 00022128 be880b24 000114a0 b6ed7e8c 20070010 00000004 00000000 00000000
[<bf004cc8>] (axidma_ioctl [xilinx_axidma]) from [<c01dcc84>] (vfs_ioctl+0x28/0x3c)
[<c01dcc84>] (vfs_ioctl) from [<c01dce28>] (do_vfs_ioctl+0x84/0x898)
[<c01dce28>] (do_vfs_ioctl) from [<c01dd670>] (SyS_ioctl+0x34/0x58)
[<c01dd670>] (SyS_ioctl) from [<c0106c60>] (ret_fast_syscall+0x0/0x3c)
Code: e3a01018 eafffead e1a0300d e0077003 (e5941008) 
---[ end trace bf7c56d3b22f2c8d ]---
```
Any idea how to resolve that?

dhzzy88, 2020-06-27:
I  have the problem,too.

ajaixavier, 2020-11-19:
This is a 'page domain fault' crash from kernel.
To avoid this, check the status of 'CONFIG_CPU_SW_DOMAIN_PAN' flag in your kernel config.
$ zcat /proc/config.gz | grep CONFIG_CPU_SW_DOMAIN_PAN
If it is in enabled state, then disable it to avoid 'page domain fault' crash from kernel.

zskdan, 2021-03-17:
Thanks @ajaixavier for your feedback. I've just finally got time to go back to this project.

 I have no more kernel panic after unsetting CONFIG_CPU_SW_DOMAIN_PAN.

`axidma_display_image` seems to be running OK. but unfortunately nothing show off from the HDMI. Do you have any working setup for this ?

Thanks!

Yuxin-Yu, 2024-08-22:
Hello @zskdan @dhzzy88 @ajaixavier , I encountered an error at the same location, but the error message is different. What could be the reason for this?
```
[ 1299.997142] Unable to handle kernel access to user memory without uaccess routines at virtual address 0000003fe89688a0
[ 1300.007734] Oops [#1]
[ 1300.009180] Modules linked in: axidma(O)
[ 1300.013100] CPU: 0 PID: 809 Comm: axidma_display_ Tainted: G           O       6.3.12-dirty #6
[ 1300.021664] Hardware name: freechips,rocketchip-vivado (DT)
[ 1300.027222] epc : axidma_ioctl+0x8e8/0xa26 [axidma]
[ 1300.039412]  ra : axidma_ioctl+0x8e8/0xa26 [axidma]
[ 1300.051540] epc : ffffffff01334cba ra : ffffffff01334cba sp : ffffffd8844ffe00
[ 1300.058700]  gp : ffffffff812e3868 tp : ffffffd88747f480 t0 : 6100000000000000
[ 1300.065900]  t1 : 0000000000000061 t2 : 61203a616d646978 s0 : ffffffd8844ffea0
[ 1300.073100]  s1 : 0000003fe8968898 a0 : 0000000000000036 a1 : ffffffd8fc5e0588
[ 1300.080306]  a2 : ffffffd8fc5e3620 a3 : 0000000000000000 a4 : d1be0e6e08481a00
[ 1300.087510]  a5 : d1be0e6e08481a00 a6 : c0000000ffffefff a7 : ffffffff81319700
[ 1300.094722]  s2 : ffffffd8844ffe10 s3 : ffffffff812e40a0 s4 : 0000000000000008
[ 1300.101920]  s5 : ffffffd883869400 s6 : 0000002ad3a36980 s7 : 0000003f8c1dfd10
[ 1300.109128]  s8 : 0000003f8c1e0008 s9 : 0000000000000000 s10: 0000002ad3a03a44
[ 1300.116334]  s11: 0000002ad3a039b8 t3 : ffffffff812f589f t4 : ffffffff812f589f
[ 1300.123536]  t5 : ffffffff812f58a0 t6 : ffffffd8844ffc28
[ 1300.128828] status: 0000000200000120 badaddr: 0000003fe89688a0 cause: 000000000000000d
[ 1300.136766] [<ffffffff01334cba>] axidma_ioctl+0x8e8/0xa26 [axidma]
[ 1300.150258] [<ffffffff80181b08>] sys_ioctl+0x6e/0x8c
[ 1300.155172] [<ffffffff80003570>] ret_from_syscall+0x0/0x2
[ 1300.160760] Code: 0613 6066 3517 0000 0513 3c65 9097 7f38 80e7 ab00 (6488) 85d2 
[ 1300.169406] ---[ end trace 0000000000000000 ]---
Segmentation fault
```