bperez77/xilinx_axidma #123: Can not build petalinux module

loylosky, 2020-11-03:
Hi,
Thank you for your kind sharing!! 
After I created module and finished the steps as guidence, I just cannot build the module and the error information as follows:
`
[INFO] building project
[INFO] sourcing bitbake
[INFO] generating user layers
[INFO] generating workspace directory
INFO: bitbake petalinux-user-image
Loading cache: 100% |############################################| Time: 0:00:00
Loaded 3979 entries from dependency cache.
Parsing recipes: 100% |##########################################| Time: 0:00:04
Parsing of 2894 .bb files complete (2890 cached, 4 parsed). 3981 targets, 169 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
WARNING: /home/lullaby/pkg/petalinux/2019.2/components/yocto/source/arm/layers/meta-xilinx/meta-xilinx-bsp/recipes-kernel/linux/linux-xlnx_2019.2.bb.do_compile is tainted from a forced run
Initialising tasks: 100% |#######################################| Time: 0:00:03
Checking sstate mirror object availability: 100% |###############| Time: 0:16:55
Sstate summary: Wanted 856 Found 676 Missed 360 Current 35 (78% match, 79% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
NOTE: linux-xlnx: compiling from external source tree /home/lullaby/work/axidma/components/plnx_workspace/sources/linux-xlnx
ERROR: axidma-1.0-r0 do_compile: oe_runmake failed
ERROR: axidma-1.0-r0 do_compile: Function failed: do_compile (log file is located at /home/lullaby/work/axidma/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/axidma/1.0-r0/temp/log.do_compile.45222)
ERROR: Logfile of failure stored in: /home/lullaby/work/axidma/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/axidma/1.0-r0/temp/log.do_compile.45222
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 4 KERNEL_SRC=/home/lullaby/work/axidma/build/tmp/work-shared/plnx-zynq7/kernel-source KERNEL_PATH=/home/lullaby/work/axidma/build/tmp/work-shared/plnx-zynq7/kernel-source KERNEL_VERSION=4.19.0-xilinx-v2019.2 CC=arm-xilinx-linux-gnueabi-gcc  -mno-thumb-interwork -marm -fuse-ld=bfd -fdebug-prefix-map=/home/lullaby/work/axidma/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/axidma/1.0-r0=/usr/src/debug/axidma/1.0-r0 -fdebug-prefix-map=/home/lullaby/work/axidma/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/axidma/1.0-r0/recipe-sysroot= -fdebug-prefix-map=/home/lullaby/work/axidma/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/axidma/1.0-r0/recipe-sysroot-native=  -fdebug-prefix-map=/home/lullaby/work/axidma/build/tmp/work-shared/plnx-zynq7/kernel-source=/usr/src/kernel LD=arm-xilinx-linux-gnueabi-ld.bfd   AR=arm-xilinx-linux-gnueabi-ar  O=/home/lullaby/work/axidma/build/tmp/work-shared/plnx-zynq7/kernel-build-artifacts KBUILD_EXTRA_SYMBOLS=
| make -C /home/lullaby/work/axidma/build/tmp/work-shared/plnx-zynq7/kernel-source M=/home/lullaby/work/axidma/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/axidma/1.0-r0
| make[1]: Entering directory '/home/lullaby/work/axidma/components/plnx_workspace/sources/linux-xlnx'
| make[2]: Entering directory '/home/lullaby/work/axidma/build/tmp/work-shared/plnx-zynq7/kernel-build-artifacts'
|   CC [M]  /home/lullaby/work/axidma/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/axidma/1.0-r0/axi_dma.o
|   CC [M]  /home/lullaby/work/axidma/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/axidma/1.0-r0/axidma_dma.o
|   CC [M]  /home/lullaby/work/axidma/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/axidma/1.0-r0/axidma_chrdev.o
|   CC [M]  /home/lullaby/work/axidma/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/axidma/1.0-r0/axidma_of.o
| /home/lullaby/work/axidma/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/axidma/1.0-r0/axi_dma.c:97:34: error: array type has incomplete element type 'struct of_device_id'
|  static const struct of_device_id axidma_compatible_of_ids[] = {
|                                   ^~~~~~~~~~~~~~~~~~~~~~~~
| /home/lullaby/work/axidma/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/axidma/1.0-r0/axi_dma.c:98:7: error: field name not in record or union initializer
|      { .compatible = "xlnx,axidma-chrdev" },
|        ^
| /home/lullaby/work/axidma/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/axidma/1.0-r0/axi_dma.c:98:7: note: (near initialization for 'axidma_compatible_of_ids')
| /home/lullaby/work/axidma/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/axidma/1.0-r0/axi_dma.c:97:34: warning: 'axidma_compatible_of_ids' defined but not used [-Wunused-variable]
|  static const struct of_device_id axidma_compatible_of_ids[] = {
|                                   ^~~~~~~~~~~~~~~~~~~~~~~~
| make[3]: *** [/home/lullaby/work/axidma/components/plnx_workspace/sources/linux-xlnx/scripts/Makefile.build:306: /home/lullaby/work/axidma/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/axidma/1.0-r0/axi_dma.o] Error 1
| make[3]: *** Waiting for unfinished jobs....
| /home/lullaby/work/axidma/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/axidma/1.0-r0/axidma_chrdev.c: In function 'axidma_mmap':
| /home/lullaby/work/axidma/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/axidma/1.0-r0/axidma_chrdev.c:277:5: error: too few arguments to function 'of_dma_configure'
|      of_dma_configure(dev->device, NULL);
|      ^~~~~~~~~~~~~~~~
| In file included from /home/lullaby/work/axidma/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/axidma/1.0-r0/axidma_chrdev.c:24:
| /home/lullaby/work/axidma/components/plnx_workspace/sources/linux-xlnx/include/linux/of_device.h:58:5: note: declared here
|  int of_dma_configure(struct device *dev,
|      ^~~~~~~~~~~~~~~~
| make[3]: *** [/home/lullaby/work/axidma/components/plnx_workspace/sources/linux-xlnx/scripts/Makefile.build:306: /home/lullaby/work/axidma/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/axidma/1.0-r0/axidma_chrdev.o] Error 1
| ERROR: oe_runmake failed
| make[2]: *** [/home/lullaby/work/axidma/components/plnx_workspace/sources/linux-xlnx/Makefile:1517: _module_/home/lullaby/work/axidma/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/axidma/1.0-r0] Error 2
| make[2]: Leaving directory '/home/lullaby/work/axidma/build/tmp/work-shared/plnx-zynq7/kernel-build-artifacts'
| make[1]: *** [Makefile:146: sub-make] Error 2
| make[1]: Leaving directory '/home/lullaby/work/axidma/components/plnx_workspace/sources/linux-xlnx'
| make: *** [Makefile:8: all] Error 2
| WARNING: /home/lullaby/work/axidma/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/axidma/1.0-r0/temp/run.do_compile.45222:1 exit 1 from 'exit 1'
| ERROR: Function failed: do_compile (log file is located at /home/lullaby/work/axidma/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/axidma/1.0-r0/temp/log.do_compile.45222)
ERROR: Task (/home/lullaby/work/axidma/project-spec/meta-user/recipes-modules/axidma/axidma.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3198 tasks of which 2364 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/lullaby/work/axidma/project-spec/meta-user/recipes-modules/axidma/axidma.bb:do_compile
Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
ERROR: Failed to build project
`
Does anyone encounter the same error? I would really appreciate it if anyone can provide some suggestions! Thanks a lot!!

Best regards!

loylosky, 2020-11-05:
I have solved the issue, just 1)add "#include <linux / of_address.h>" into driver / axi_dma.c
and 2)in axidma_chrdev.c, change the parameter from "of_dma_configure(dev->device, NULL,) ;" to "of_dma_configure(dev->device, NULL, true) ;" Hope this can help. ^_^

lordlothard, 2021-05-12:
Hi  ! I had the same problem  in **May, 2021** . I use linux-xlnx-xilinx-v2019.2.01 
Your workaround worked well ! (at least it builds).  In case people face the same issue i allow myself to present the issue clearly : 

**Issue :**  
```
$ make driver
make -C /<your_linux_path>/linux-xlnx-xilinx-v2019.2.01 M=/<your_driver_path>/xilinx_axidma-master/driver modules
make[1]: Entering directory `/<your_linux_path>/linux-xlnx-xilinx-v2019.2.01'
  CC [M]  /<your_driver_path>/xilinx_axidma-master/driver/axi_dma.o
/<your_driver_path>/xilinx_axidma-master/driver/axi_dma.c:97:34: error: array type has incomplete element type ‘struct of_device_id’
 static const struct of_device_id axidma_compatible_of_ids[] = {
                                  ^~~~~~~~~~~~~~~~~~~~~~~~
/<your_driver_path>/xilinx_axidma-master/driver/axi_dma.c:98:7: error: field name not in record or union initializer
     { .compatible = "xlnx,axidma-chrdev" },
       ^
/<your_driver_path>/xilinx_axidma-master/driver/axi_dma.c:98:7: note: (near initialization for ‘axidma_compatible_of_ids’)
/<your_driver_path>/xilinx_axidma-master/driver/axi_dma.c:97:34: error: ‘axidma_compatible_of_ids’ defined but not used [-Werror=unused-variable]
 static const struct of_device_id axidma_compatible_of_ids[] = {
                                  ^~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [/<your_driver_path>/xilinx_axidma-master/driver/axi_dma.o] Error 1
make[1]: *** [_module_/<your_driver_path>/xilinx_axidma-master/driver] Error 2
make[1]: Leaving directory `/<your_linux_path>/linux-xlnx-xilinx-v2019.2.01'
make: *** [driver/axidma.ko] Error 2
```

**Workaroud**

1. add "`#include <linux / of_address.h>`" into driver / **_axi_dma.c_**
2. in _**axidma_chrdev.c**_, change the parameter from "`of_dma_configure(dev->device, NULL,) ;`" to "`of_dma_configure(dev->device, NULL, true) ;`"

> I have solved the issue, just 1)add "#include <linux / of_address.h>" into driver / axi_dma.c
> and 2)in axidma_chrdev.c, change the parameter from "of_dma_configure(dev->device, NULL,) ;" to "of_dma_configure(dev->device, NULL, true) ;" Hope this can help. ^_^

Thank you for your help ! Maybe this can be integrated in a pull request ?

Iyiren, 2022-05-04:
I use petalinux 2021.1 build the module, there is two errors. axidma_dma.c: void axidma_dma_callback() func, struct siginfo error, new is struct kernel_siginfo. axidma_chrdev.c: access_ok() only has two arguments.

qxtcreat, 2025-01-16:
@Iyiren hello! Did you have solve these problem? I also want to use this project with petalinux 2021.1.But I have noticed that this driver supports 4.x version Xilinx kernels.However,The petalinux 2021.1 is 5.10. If you have solved those,would you be willing to share more information.Thank you very much.

rureverek, 2025-03-24:
Use: https://github.com/bperez77/xilinx_axidma/pull/139 and append `MODULE_IMPORT_NS(DMA_BUF);` to the end of `axi_dma.c`, just before MODULE_AUTHOR()