bperez77/xilinx_axidma #125: Multichannel DMA transfer error

Vanau, 2020-11-30:
Hi,I am trying to use your case for my testing procedure.But currently some bugs were found and puzzled me a lot.
my dts can be listed as
`/ {
	amba_pl: amba_pl {
		#address-cells = <1>;
		#size-cells = <1>;
		compatible = "simple-bus";
		ranges ;
		
		axidma_chrdev@0{
			compatible = "xlnx,axidma-chrdev";
			dmas = <&axi_dma_0 0 &axi_dma_1 0 &axi_dma_1 1>;
			dma-names = "rx_channel_test", "tx_channel", "rx_channel";
		};

		axi_dma_0: dma@40400000 {
			#dma-cells = <1>;
			clock-names = "s_axi_lite_aclk", "m_axi_sg_aclk", "m_axi_s2mm_aclk";
			clocks = <&clkc 15>, <&clkc 15>, <&clkc 15>;
			compatible = "xlnx,axi-dma-1.00.a";
			interrupt-parent = <&intc>;
			interrupts = <0 29 4>;
			reg = <0x40400000 0x10000>;
			xlnx,addrwidth = <0x20>;
			// xlnx,include-sg;
			// xlnx,sg-length-width = <0x10>;
			dma-channel@40400030 {
				compatible = "xlnx,axi-dma-s2mm-channel";
				dma-channels = <0x1>;
				interrupts = <0 29 4>;
				xlnx,datawidth = <0x10>;
				xlnx,device-id = <0x0>;
			};
		};
		axi_dma_1: dma@40410000 {
			#dma-cells = <1>;
			clock-names = "s_axi_lite_aclk", "m_axi_sg_aclk", "m_axi_mm2s_aclk", "m_axi_s2mm_aclk";
			clocks = <&clkc 15>, <&clkc 15>, <&clkc 15>, <&clkc 15>;
			compatible = "xlnx,axi-dma-1.00.a";
			interrupt-parent = <&intc>;
			interrupts = <0 30 4 0 31 4>;
			reg = <0x40410000 0x10000>;
			xlnx,addrwidth = <0x20>;
			// xlnx,include-sg;
			// xlnx,sg-length-width = <0x10>;
			dma-channel@40410000 {
				compatible = "xlnx,axi-dma-mm2s-channel";
				dma-channels = <0x1>;
				interrupts = <0 30 4>;
				xlnx,datawidth = <0x20>;
				xlnx,device-id = <0x1>;
			};
			dma-channel@40410030 {
				compatible = "xlnx,axi-dma-s2mm-channel";
				dma-channels = <0x1>;
				interrupts = <0 31 4>;
				xlnx,datawidth = <0x20>;
				xlnx,device-id = <0x2>;
			};
		};
	};
};`

Vanau, 2020-11-30:
The question is:
1.I didn't use sg mode (as shown in the device tree);
2.I followed  this order with the author: axidma_init() -> axidma_get_dma_tx() -> axidma_oneway_transfer();
It is right to use a DMA which has only one send channel,as well as both send and receive channels together.
But when I use two DMAs, this will give me an error.
The problem puzzled me a few days,and any advice would be appreciated.Thanks!

Hanxin-1024, 2021-04-15:
I also encountered a similar problem. I mounted 4 channels of DMA at the same time, each with a receiving channel and a sending channel. There was no problem when testing a channel of DMA separately, but I encountered the same error when running 8 channels simultaneously``

Hanxin-1024, 2021-04-15:
I also encountered a similar problem. I mounted 4 AXIDMA at the same time, each with a receiving channel and a sending channel. There was no problem when testing a channel of DMA separately, but I encountered the same error when running 8 channels simultaneously

Hanxin-1024, 2021-04-16:
> 我也遇到了类似的问题。我同时安装了4个AXIDMA，每个都有一个接收通道和一个发送通道。分别测试DMA通道时没有问题，但是同时运行8个通道时遇到了相同的错误

No problem now，can support multiple channels open at the same time

Hanxin-1024, 2021-04-16:
> The question is:
> 1.I didn't use sg mode (as shown in the device tree);
> 2.I followed this order with the author: axidma_init() -> axidma_get_dma_tx() -> axidma_oneway_transfer();
> It is right to use a [...]

You can try mmap different spaces

hxl-make, 2022-03-10:
> > 问题是：
> > 1.我没有使用sg模式（如设备树所示）；
> > 2.我跟作者的顺序是这样的：axidma_init() -> axidma_get_dma_tx() -> axidma_oneway_transfer(); 
> > 使用只有一个发送通道的 DMA 是正确的，发送通道和接收通道一起使用。
> > 但是当我使用两个 DMA 时，这会给我一个错误。
> > 这个问题困扰了我几天，任何建议都将不胜感激。谢谢！
> 
> 您可以尝试 mmap 不同的空间

Hello, can you refer to your modification? I'm a novice

Hanxin-1024, 2022-03-10:
https://blog.csdn.net/qq_42330920/article/details/115618482?spm=1001.2014.3001.5501
https://blog.csdn.net/qq_42330920/article/details/115704028?spm=1001.2014.3001.5501
这两个博客是我当时的调试记录，有更改思路以及我最终的修改办法，参考下看有没有帮助呢

   

 2.我跟作者的顺序是这样的：axidma_init() -&gt; axidma_get_dma_tx() -&gt; axidma_oneway_transfer();