bperez77/xilinx_axidma #127: Build problem with petalinux 20.2

HosseinBeheshti, 2021-01-27:
Hi
I am using Petalinux 20.2 to build Linux for my custom project.
In the new version of the kernel, some functions are changed and I get these errors when running `petalinux-build`:

````console
ERROR: xilinx-axidma-1.0-r0 do_compile: oe_runmake failed
ERROR: xilinx-axidma-1.0-r0 do_compile: Execution of '/media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/build/tmp/work/zynqmp_generic-xilinx-linux/xilinx-axidma/1.0-r0/temp/run.do_compile.13695' failed with exit code 1:
make -C /media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/build/tmp/work-shared/zynqmp-generic/kernel-source M=/media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/build/tmp/work/zynqmp_generic-xilinx-linux/xilinx-axidma/1.0-r0
make[1]: Entering directory '/media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/components/yocto/workspace/sources/linux-xlnx'
make[2]: Entering directory '/media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/build/tmp/work-shared/zynqmp-generic/kernel-build-artifacts'
  AR      /media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/build/tmp/work/zynqmp_generic-xilinx-linux/xilinx-axidma/1.0-r0/built-in.a
  CC [M]  /media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/build/tmp/work/zynqmp_generic-xilinx-linux/xilinx-axidma/1.0-r0/axidma_chrdev.o
  CC [M]  /media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/build/tmp/work/zynqmp_generic-xilinx-linux/xilinx-axidma/1.0-r0/axi_dma.o
  CC [M]  /media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/build/tmp/work/zynqmp_generic-xilinx-linux/xilinx-axidma/1.0-r0/axidma_dma.o
  CC [M]  /media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/build/tmp/work/zynqmp_generic-xilinx-linux/xilinx-axidma/1.0-r0/axidma_of.o
/media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/build/tmp/work/zynqmp_generic-xilinx-linux/xilinx-axidma/1.0-r0/axidma_dma.c: In function 'axidma_dma_callback':
/media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/build/tmp/work/zynqmp_generic-xilinx-linux/xilinx-axidma/1.0-r0/axidma_dma.c:158:47: error: passing argument 2 of 'send_sig_info' from incompatible pointer type [-Werror=incompatible-pointer-types]
  158 |         send_sig_info(cb_data->notify_signal, &sig_info, cb_data->process);
      |                                               ^~~~~~~~~
      |                                               |
      |                                               struct siginfo *
In file included from /media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/build/tmp/work/zynqmp_generic-xilinx-linux/xilinx-axidma/1.0-r0/axidma_dma.c:21:
/media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/components/yocto/workspace/sources/linux-xlnx/include/linux/sched/signal.h:322:31: note: expected 'struct kernel_siginfo *' but argument is of type 'struct siginfo *'
  322 | extern int send_sig_info(int, struct kernel_siginfo *, struct task_struct *);
      |                               ^~~~~~~~~~~~~~~~~~~~~~~
/media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/build/tmp/work/zynqmp_generic-xilinx-linux/xilinx-axidma/1.0-r0/axidma_chrdev.c: In function 'axidma_access_ok':
/media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/build/tmp/work/zynqmp_generic-xilinx-linux/xilinx-axidma/1.0-r0/axidma_chrdev.c:329:56: error: macro "access_ok" passed 3 arguments, but takes just 2
  329 |     if (!readonly && !access_ok(VERIFY_WRITE, arg, size)) {
      |                                                        ^
In file included from /media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/components/yocto/workspace/sources/linux-xlnx/include/linux/uaccess.h:11,
                 from /media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/build/tmp/work/zynqmp_generic-xilinx-linux/xilinx-axidma/1.0-r0/axidma_chrdev.c:21:
/media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/components/yocto/workspace/sources/linux-xlnx/arch/arm64/include/asm/uaccess.h:90: note: macro "access_ok" defined here
   90 | #define access_ok(addr, size) __range_ok(addr, size)
      | 
/media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/build/tmp/work/zynqmp_generic-xilinx-linux/xilinx-axidma/1.0-r0/axidma_chrdev.c:329:23: error: 'access_ok' undeclared (first use in this function)
  329 |     if (!readonly && !access_ok(VERIFY_WRITE, arg, size)) {
      |                       ^~~~~~~~~
/media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/build/tmp/work/zynqmp_generic-xilinx-linux/xilinx-axidma/1.0-r0/axidma_chrdev.c:329:23: note: each undeclared identifier is reported only once for each function it appears in
/media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/build/tmp/work/zynqmp_generic-xilinx-linux/xilinx-axidma/1.0-r0/axidma_chrdev.c:333:49: error: macro "access_ok" passed 3 arguments, but takes just 2
  333 |     } else if (!access_ok(VERIFY_READ, arg, size)) {
      |                                                 ^
In file included from /media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/components/yocto/workspace/sources/linux-xlnx/include/linux/uaccess.h:11,
                 from /media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/build/tmp/work/zynqmp_generic-xilinx-linux/xilinx-axidma/1.0-r0/axidma_chrdev.c:21:
/media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/components/yocto/workspace/sources/linux-xlnx/arch/arm64/include/asm/uaccess.h:90: note: macro "access_ok" defined here
   90 | #define access_ok(addr, size) __range_ok(addr, size)
      | 
make[3]: *** [/media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/components/yocto/workspace/sources/linux-xlnx/scripts/Makefile.build:266: /media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/build/tmp/work/zynqmp_generic-xilinx-linux/xilinx-axidma/1.0-r0/axidma_chrdev.o] Error 1
make[3]: *** Waiting for unfinished jobs....
cc1: some warnings being treated as errors
make[3]: *** [/media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/components/yocto/workspace/sources/linux-xlnx/scripts/Makefile.build:265: /media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/build/tmp/work/zynqmp_generic-xilinx-linux/xilinx-axidma/1.0-r0/axidma_dma.o] Error 1
make[2]: *** [/media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/components/yocto/workspace/sources/linux-xlnx/Makefile:1652: /media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/build/tmp/work/zynqmp_generic-xilinx-linux/xilinx-axidma/1.0-r0] Error 2
make[2]: Leaving directory '/media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/build/tmp/work-shared/zynqmp-generic/kernel-build-artifacts'
make[1]: *** [Makefile:179: sub-make] Error 2
make[1]: Leaving directory '/media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/components/yocto/workspace/sources/linux-xlnx'
make: *** [Makefile:11: all] Error 2
WARNING: /media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/build/tmp/work/zynqmp_generic-xilinx-linux/xilinx-axidma/1.0-r0/temp/run.do_compile.13695:1 exit 1 from 'exit 1'

ERROR: Logfile of failure stored in: /media/fpgaws/projects/DevelopmentLab/build/apu/ultrazed7ev/build/tmp/work/zynqmp_generic-xilinx-linux/xilinx-axidma/1.0-r0/temp/log.do_compile.13695
````
I change the contents of the corresponding file (axi_dma.c, axidma_chrdev.c and axidma_dma.c) as following:
axi_dma.c (add in line 18):
````c
#include <linux/of_address.h>
````
axidma_chrdev.c (change in line 277):
````c
of_dma_configure(dev->device, NULL, true);
````
axi_dma.c (chang in line 146):
````c
    struct kernel_siginfo sig_info;
````
Do these changes are right?

ZLXT, 2021-07-11:
Hello, I met the same problem and used the solution you provided. But, there still  some errors  at compile time.
Like this:

```
ERROR: xilinx-axidma-1.0-r0 do_compile: oe_runmake failed
ERROR: xilinx-axidma-1.0-r0 do_compile: Execution of '/home/zlxt/Project/Petalinux/SG_DMA_Z020/SG_DMA/build/tmp/work/zynq_generic-xilinx-linux-gnueabi/xilinx-axidma/1.0-r0/temp/run.do_compile.5325' failed with exit code 1:
make -C /home/zlxt/Project/Petalinux/SG_DMA_Z020/SG_DMA/build/tmp/work-shared/zynq-generic/kernel-source M=/home/zlxt/Project/Petalinux/SG_DMA_Z020/SG_DMA/build/tmp/work/zynq_generic-xilinx-linux-gnueabi/xilinx-axidma/1.0-r0
make[1]: Entering directory '/home/zlxt/Project/Petalinux/linux-xlnx-xilinx-v2020.2.2-k26'
make[2]: Entering directory '/home/zlxt/Project/Petalinux/SG_DMA_Z020/SG_DMA/build/tmp/work-shared/zynq-generic/kernel-build-artifacts'
  AR      /home/zlxt/Project/Petalinux/SG_DMA_Z020/SG_DMA/build/tmp/work/zynq_generic-xilinx-linux-gnueabi/xilinx-axidma/1.0-r0/built-in.a
  CC [M]  /home/zlxt/Project/Petalinux/SG_DMA_Z020/SG_DMA/build/tmp/work/zynq_generic-xilinx-linux-gnueabi/xilinx-axidma/1.0-r0/axi_dma.o
  CC [M]  /home/zlxt/Project/Petalinux/SG_DMA_Z020/SG_DMA/build/tmp/work/zynq_generic-xilinx-linux-gnueabi/xilinx-axidma/1.0-r0/axidma_chrdev.o
  CC [M]  /home/zlxt/Project/Petalinux/SG_DMA_Z020/SG_DMA/build/tmp/work/zynq_generic-xilinx-linux-gnueabi/xilinx-axidma/1.0-r0/axidma_dma.o
  CC [M]  /home/zlxt/Project/Petalinux/SG_DMA_Z020/SG_DMA/build/tmp/work/zynq_generic-xilinx-linux-gnueabi/xilinx-axidma/1.0-r0/axidma_of.o
/home/zlxt/Project/Petalinux/SG_DMA_Z020/SG_DMA/build/tmp/work/zynq_generic-xilinx-linux-gnueabi/xilinx-axidma/1.0-r0/axidma_chrdev.c: In function 'axidma_access_ok':
/home/zlxt/Project/Petalinux/SG_DMA_Z020/SG_DMA/build/tmp/work/zynq_generic-xilinx-linux-gnueabi/xilinx-axidma/1.0-r0/axidma_chrdev.c:329:56: error: macro "access_ok" passed 3 arguments, but takes just 2
  329 |     if (!readonly && !access_ok(VERIFY_WRITE, arg, size)) {
      |                                                        ^
In file included from /home/zlxt/Project/Petalinux/linux-xlnx-xilinx-v2020.2.2-k26/include/linux/uaccess.h:11,
                 from /home/zlxt/Project/Petalinux/SG_DMA_Z020/SG_DMA/build/tmp/work/zynq_generic-xilinx-linux-gnueabi/xilinx-axidma/1.0-r0/axidma_chrdev.c:21:
/home/zlxt/Project/Petalinux/linux-xlnx-xilinx-v2020.2.2-k26/arch/arm/include/asm/uaccess.h:280: note: macro "access_ok" defined here
  280 | #define access_ok(addr, size) (__range_ok(addr, size) == 0)
      | 
/home/zlxt/Project/Petalinux/SG_DMA_Z020/SG_DMA/build/tmp/work/zynq_generic-xilinx-linux-gnueabi/xilinx-axidma/1.0-r0/axidma_chrdev.c:329:23: error: 'access_ok' undeclared (first use in this function)
  329 |     if (!readonly && !access_ok(VERIFY_WRITE, arg, size)) {
      |                       ^~~~~~~~~
/home/zlxt/Project/Petalinux/SG_DMA_Z020/SG_DMA/build/tmp/work/zynq_generic-xilinx-linux-gnueabi/xilinx-axidma/1.0-r0/axidma_chrdev.c:329:23: note: each undeclared identifier is reported only once for each function it appears in
/home/zlxt/Project/Petalinux/SG_DMA_Z020/SG_DMA/build/tmp/work/zynq_generic-xilinx-linux-gnueabi/xilinx-axidma/1.0-r0/axidma_chrdev.c:333:49: error: macro "access_ok" passed 3 arguments, but takes just 2
  333 |     } else if (!access_ok(VERIFY_READ, arg, size)) {
      |                                                 ^
In file included from /home/zlxt/Project/Petalinux/linux-xlnx-xilinx-v2020.2.2-k26/include/linux/uaccess.h:11,
                 from /home/zlxt/Project/Petalinux/SG_DMA_Z020/SG_DMA/build/tmp/work/zynq_generic-xilinx-linux-gnueabi/xilinx-axidma/1.0-r0/axidma_chrdev.c:21:
/home/zlxt/Project/Petalinux/linux-xlnx-xilinx-v2020.2.2-k26/arch/arm/include/asm/uaccess.h:280: note: macro "access_ok" defined here
  280 | #define access_ok(addr, size) (__range_ok(addr, size) == 0)
      | 
make[3]: *** [/home/zlxt/Project/Petalinux/linux-xlnx-xilinx-v2020.2.2-k26/scripts/Makefile.build:266: /home/zlxt/Project/Petalinux/SG_DMA_Z020/SG_DMA/build/tmp/work/zynq_generic-xilinx-linux-gnueabi/xilinx-axidma/1.0-r0/axidma_chrdev.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [/home/zlxt/Project/Petalinux/linux-xlnx-xilinx-v2020.2.2-k26/Makefile:1652: /home/zlxt/Project/Petalinux/SG_DMA_Z020/SG_DMA/build/tmp/work/zynq_generic-xilinx-linux-gnueabi/xilinx-axidma/1.0-r0] Error 2
make[2]: Leaving directory '/home/zlxt/Project/Petalinux/SG_DMA_Z020/SG_DMA/build/tmp/work-shared/zynq-generic/kernel-build-artifacts'
make[1]: *** [Makefile:179: sub-make] Error 2
make[1]: Leaving directory '/home/zlxt/Project/Petalinux/linux-xlnx-xilinx-v2020.2.2-k26'
make: *** [Makefile:10: all] Error 2
WARNING: /home/zlxt/Project/Petalinux/SG_DMA_Z020/SG_DMA/build/tmp/work/zynq_generic-xilinx-linux-gnueabi/xilinx-axidma/1.0-r0/temp/run.do_compile.5325:1 exit 1 from 'exit 1'

ERROR: Logfile of failure stored in: /home/zlxt/Project/Petalinux/SG_DMA_Z020/SG_DMA/build/tmp/work/zynq_generic-xilinx-linux-gnueabi/xilinx-axidma/1.0-r0/temp/log.do_compile.5325
```

Do you know how to fix this problem?
What causes this problem?
Thanks!  : )

HosseinBeheshti, 2021-07-11:
Hi
I do not remember any other changes in my code. But you can access the final version of my code vi [this link](https://gitlab.gnome.org/HosseinBeheshti/UlteraZed7EV/-/tree/master/petalinux/xilinx_axidma).
If you have any problems, please let me know.

ZLXT, 2021-07-11:
Thank you for your reply. 
I checked your file and found different in 'axidma_chrdev.c'file, 'access_ok'function has only two arguments. 
This corresponds to the error prompt. I think this problme caused by the adjustment of the petalinux kernel function.

axidma_chrdev.c (change in line 329):

```diff
-      if (!readonly && !access_ok(VERIFY_WRITE, arg, size)) {
+      if (!readonly && !access_ok(arg, size)) {
```
axidma_chrdev.c (change in line 333):

```diff
-     } else if (!access_ok(VERIFY_READ, arg, size)) {
+    } else if (!access_ok(VERIFY_READ, arg, size)) {
```

HosseinBeheshti, 2021-07-11:
Thank you @ZLXT

closed 2021-07-11 by HosseinBeheshti