bperez77/xilinx_axidma #132: concurrent access to DMA engines

samberhanu, 2021-02-27:
I know that there's a fork [corona](https://github.com/corna/xilinx_axidma/graphs/code-frequency) but I am having trouble compiling that correctly:
Ex: the following is when I tried to include ./linux/uaccess.h (because it kept complaining that it couldn't find copy_to_user) if asm/uaccess.h was the one in the include
In function ‘check_copy_size’,
    inlined from ‘copy_from_user’ at ./include/linux/uaccess.h:146:6,
    inlined from ‘axidma_ioctl’ at /home/berhas1/BIGDEAL/linux_bsp_demo2/zcu10x_bsp/xilinx_axidma/driver/axidma_chrdev.c:389:17:
./include/linux/thread_info.h:141:4: error: call to ‘__bad_copy_to’ declared with attribute error: copy destination size is too small
    __bad_copy_to();

The short of it: is this driver able to handle multiple dma engines concurrently. I am ok with the Tx/Rx pairs being dictated on only one process but would really love to have the DMAs calls running concurrently

Yuxin-Yu, 2024-07-05:
Hi bro @samberhanu ,have you solve this problem?

wangzhenhua92429, 2024-07-05:
你发给我的邮件已收到！                                ——王振华