bperez77/xilinx_axidma #136: Can't build driver on petalinux 2020.2

SzateX, 2021-08-02:
Hi. I'm trying to compile driver for petalinux 2020.2. But I got his message:
`
NOTE: Executing Tasks
NOTE: Setscene tasks completed
ERROR: xilinix-axidma-1.0-r0 do_compile: oe_runmake failed
ERROR: xilinix-axidma-1.0-r0 do_compile: Execution of '/home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/temp/run.do_compile.18889' failed with exit code 1:
make -C /home/jszatkowski/Testowy2/build/tmp/work-shared/zynqmp-generic/kernel-source M=/home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0
make[1]: Entering directory '/home/jszatkowski/Testowy2/components/yocto/workspace/sources/linux-xlnx'
make[2]: Entering directory '/home/jszatkowski/Testowy2/build/tmp/work-shared/zynqmp-generic/kernel-build-artifacts'
  AR      /home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/built-in.a
  CC [M]  /home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/axi_dma.o
  CC [M]  /home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/axidma_chrdev.o
  CC [M]  /home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/axidma_of.o
  CC [M]  /home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/axidma_dma.o
/home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/axidma_dma.c: In function 'axidma_dma_callback':
/home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/axidma_dma.c:158:47: error: passing argument 2 of 'send_sig_info' from incompatible pointer type [-Werror=incompatible-pointer-types]
  158 |         send_sig_info(cb_data->notify_signal, &sig_info, cb_data->process);
      |                                               ^~~~~~~~~
      |                                               |
      |                                               struct siginfo *
In file included from /home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/axidma_dma.c:21:
/home/jszatkowski/Testowy2/components/yocto/workspace/sources/linux-xlnx/include/linux/sched/signal.h:322:31: note: expected 'struct kernel_siginfo *' but argument is of type 'struct siginfo *'
  322 | extern int send_sig_info(int, struct kernel_siginfo *, struct task_struct *);
      |                               ^~~~~~~~~~~~~~~~~~~~~~~
/home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/axidma_chrdev.c: In function 'axidma_access_ok':
/home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/axidma_chrdev.c:329:56: error: macro "access_ok" passed 3 arguments, but takes just 2
  329 |     if (!readonly && !access_ok(VERIFY_WRITE, arg, size)) {
      |                                                        ^
In file included from /home/jszatkowski/Testowy2/components/yocto/workspace/sources/linux-xlnx/include/linux/uaccess.h:11,
                 from /home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/axidma_chrdev.c:21:
/home/jszatkowski/Testowy2/components/yocto/workspace/sources/linux-xlnx/arch/arm64/include/asm/uaccess.h:90: note: macro "access_ok" defined here
   90 | #define access_ok(addr, size) __range_ok(addr, size)
      |
/home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/axidma_chrdev.c:329:23: error: 'access_ok' undeclared (first use in this function)
  329 |     if (!readonly && !access_ok(VERIFY_WRITE, arg, size)) {
      |                       ^~~~~~~~~
/home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/axidma_chrdev.c:329:23: note: each undeclared identifier is reported only once for each function it appears in
/home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/axidma_chrdev.c:333:49: error: macro "access_ok" passed 3 arguments, but takes just 2
  333 |     } else if (!access_ok(VERIFY_READ, arg, size)) {
      |                                                 ^
In file included from /home/jszatkowski/Testowy2/components/yocto/workspace/sources/linux-xlnx/include/linux/uaccess.h:11,
                 from /home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/axidma_chrdev.c:21:
/home/jszatkowski/Testowy2/components/yocto/workspace/sources/linux-xlnx/arch/arm64/include/asm/uaccess.h:90: note: macro "access_ok" defined here
   90 | #define access_ok(addr, size) __range_ok(addr, size)
      |
make[3]: *** [/home/jszatkowski/Testowy2/components/yocto/workspace/sources/linux-xlnx/scripts/Makefile.build:266: /home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/axidma_chrdev.o] Error 1
make[3]: *** Waiting for unfinished jobs....
cc1: some warnings being treated as errors
make[3]: *** [/home/jszatkowski/Testowy2/components/yocto/workspace/sources/linux-xlnx/scripts/Makefile.build:265: /home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/axidma_dma.o] Error 1
make[2]: *** [/home/jszatkowski/Testowy2/components/yocto/workspace/sources/linux-xlnx/Makefile:1652: /home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0] Error 2
make[2]: Leaving directory '/home/jszatkowski/Testowy2/build/tmp/work-shared/zynqmp-generic/kernel-build-artifacts'
make[1]: *** [Makefile:179: sub-make] Error 2
make[1]: Leaving directory '/home/jszatkowski/Testowy2/components/yocto/workspace/sources/linux-xlnx'
make: *** [Makefile:11: all] Error 2
WARNING: /home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/temp/run.do_compile.18889:1 exit 1 from 'exit 1'

ERROR: Logfile of failure stored in: /home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/temp/log.do_compile.18889
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 48 KERNEL_SRC=/home/jszatkowski/Testowy2/build/tmp/work-shared/zynqmp-generic/kernel-source KERNEL_PATH=/home/jszatkowski/Testowy2/build/tmp/work-shared/zynqmp-generic/kernel-source KERNEL_VERSION=5.4.0-xilinx-v2020.2 CC=aarch64-xilinx-linux-gcc   -fuse-ld=bfd -fmacro-prefix-map=/home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0=/usr/src/debug/xilinix-axidma/1.0-r0                      -fdebug-prefix-map=/home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0=/usr/src/debug/xilinix-axidma/1.0-r0                      -fdebug-prefix-map=/home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/recipe-sysroot-native=  -fdebug-prefix-map=/home/jszatkowski/Testowy2/build/tmp/work-shared/zynqmp-generic/kernel-source=/usr/src/kernel LD=aarch64-xilinx-linux-ld.bfd   AR=aarch64-xilinx-linux-ar  O=/home/jszatkowski/Testowy2/build/tmp/work-shared/zynqmp-generic/kernel-build-artifacts KBUILD_EXTRA_SYMBOLS=
| make -C /home/jszatkowski/Testowy2/build/tmp/work-shared/zynqmp-generic/kernel-source M=/home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0
| make[1]: Entering directory '/home/jszatkowski/Testowy2/components/yocto/workspace/sources/linux-xlnx'
| make[2]: Entering directory '/home/jszatkowski/Testowy2/build/tmp/work-shared/zynqmp-generic/kernel-build-artifacts'
|   AR      /home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/built-in.a
|   CC [M]  /home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/axi_dma.o
|   CC [M]  /home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/axidma_chrdev.o
|   CC [M]  /home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/axidma_of.o
|   CC [M]  /home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/axidma_dma.o
| /home/jszatkowski/Testowy2/build/tmp/work/zynqmp_generic-xilinx-linux/xilinix-axidma/1.0-r0/axidma_dma.c: In function 'axidma_dma_callback':
[... cut ...]
NOTE: Tasks Summary: Attempted 3502 tasks of which 3480 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/jszatkowski/Testowy2/project-spec/meta-user/recipes-modules/xilinix-axidma/xilinix-axidma.bb:do_compile
Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
ERROR: Failed to build project
`

I applied things from this issue: [https://github.com/bperez77/xilinx_axidma/issues/123](url)

I have no idea whats going on. Could you help me to implement this driver?

ZLXT, 2021-08-04:
Check this way: https://github.com/bperez77/xilinx_axidma/issues/127.

SzateX, 2021-08-04:
Everything works correctly. Thanks

closed 2021-08-04 by SzateX