bperez77/xilinx_axidma #140: Does the driver support one channel DMA?

jimenaCabrejas, 2021-12-16:
Hello!

I'm working on a design where I will be receiving data in the PL and I'm using a DMA to pass it to the PS. The DMA only has the write channel activated. This is my Vivado design:

[image]

I'm using Petalinux 2020.2 and my pl.dtsi looks like this
```
/ {
	amba_pl: amba_pl {
		#address-cells = <1>;
		#size-cells = <1>;
		compatible = "simple-bus";
		ranges ;
		axi_dma_0: dma@40400000 {
			#dma-cells = <1>;
			clock-names = "s_axi_lite_aclk", "m_axi_sg_aclk", "m_axi_s2mm_aclk";
			clocks = <&clkc 15>, <&clkc 15>, <&clkc 15>;
			compatible = "xlnx,axi-dma-7.1", "xlnx,axi-dma-1.00.a";
			interrupt-names = "s2mm_introut";
			interrupt-parent = <&intc>;
			interrupts = <0 29 4>;
			reg = <0x40400000 0x10000>;
			xlnx,addrwidth = <0x20>;
			xlnx,include-sg ;
			xlnx,sg-length-width = <0xe>;
			dma-channel@40400030 {
				compatible = "xlnx,axi-dma-s2mm-channel";
				dma-channels = <0x1>;
				interrupts = <0 29 4>;
				xlnx,datawidth = <0x20>;
				xlnx,device-id = <0x0>;
			};
		};
	};
};
```
In my system-user.dtsi I have
```
&amba_pl{
    axidma_chrdev: axidma_chrdev@0 {
            compatible = "xlnx,axidma-chrdev";
            dmas = <&axi_dma_0 0>;
            dma-names = "rx_channel";
    };
};

```

When I try to load the driver in petalinux using insmod I get the following error
```
xilinx_axidma: loading out-of-tree module taints kernel.
axidma: axidma_dma.c: axidma_request_channels: 651: Unable to get slave channel 0: rx_channel.
axidma: probe of amba_pl:axidma_chrdev@0 failed with error -38
```

I'm I missing something in my projecto? Does the driver support one channel DMAs? Do I need to change something to make it work?

Thank you and best regards

dahunt, 2022-07-20:
Hi,

Did you ever get a solution to your problem above?  I am running into the same issue under 2022.1.
Thanks!

Timgrau, 2022-09-22:
Hey @dahunt @jimenaCabrejas, due to this issue https://github.com/bperez77/xilinx_axidma/issues/24#issuecomment-347844532 have you tried to change the device for the dma channel in axidma_chrdev node, in system-user.dtsi to 1? i.e. `dmas = <&axi_dma_0 1>;`

dahunt, 2022-09-22:
> 

Hey @Timgrau,

Yup! that was exactly the problem.  I also had to provide an device id.  I'm still a newbie at this... but here is my working user dt entry for the Rx only streaming DMA project for your reference.

/include/ "system-conf.dtsi"
/ {
};

&amba_pl {
        dma_proxy {
                compatible = "xlnx,dma_proxy";
                dmas = <&hier_DataGen_axi_dma_0 1>;
                dma-names = "dma_proxy_rx";
                dma-coherent;
        };

        hier_DataGen_axi_dma_0: dma@80020000 {
                dma-channel@80020030 {
                        xlnx,device-id = <0x1>;
                };
        };
};

Timgrau, 2022-09-22:
Thank you @dahunt for the reference. So you used

dantepayne, 2024-01-04:
> Hey @Timgrau,
>
> Yup! that was exactly the problem. I also had to provide an device id. I'm still a newbie at this... but here is my working user dt entry for the Rx only streaming DMA project for [...]

Hey @dahunt ，I meet the same problem，Have you solve it ？

dantepayne, 2024-01-04:
When I change DT Like this：
`&amba_pl {
  axidma_chrdev: axidma_chrdev@0 {
    compatible = "xlnx,axidma-chrdev";
    dmas = <&axi_dma_0 1>;
    dma-names = "rx_channel";
  };
};

&axi_dma_0 {
  dma-channel@a0000030 {
    xlnx,device-id = <0x1>;
  };
};`
The Driver collapse.Segment fault occur.
But I use this DT：
`&amba_pl {
  axidma_chrdev: axidma_chrdev@0 {
    compatible = "xlnx,axidma-chrdev";
    dmas = <&axi_dma_0 0>;
    dma-names = "rx_channel";
  };
};

&axi_dma_0 {
  dma-channel@a0000030 {
    xlnx,device-id = <0x1>;
  };
};`
The Problem is the same.
`xilinx_axidma: loading out-of-tree module taints kernel.
axidma: axidma_dma.c: axidma_request_channels: 651: Unable to get slave channel 0: rx_channel.
axidma: probe of amba_pl:axidma_chrdev@0 failed with error -38`
Here is my pl.dtsi:
`amba_pl: amba_pl@0 {
		#address-cells = <2>;
		#size-cells = <2>;
		compatible = "simple-bus";
		ranges ;
		axi_dma_0: dma@a0000000 {
			#dma-cells = <1>;
			clock-names = "s_axi_lite_aclk", "m_axi_s2mm_aclk";
			clocks = <&zynqmp_clk 71>, <&zynqmp_clk 71>;
			compatible = "xlnx,axi-dma-7.1", "xlnx,axi-dma-1.00.a";
			interrupt-names = "s2mm_introut";
			interrupt-parent = <&gic>;
			interrupts = <0 89 4>;
			reg = <0x0 0xa0000000 0x0 0x10000>;
			xlnx,addrwidth = <0x20>;
			xlnx,sg-length-width = <0x17>;
			dma-channel@a0000030 {
				compatible = "xlnx,axi-dma-s2mm-channel";
				dma-channels = <0x1>;
				interrupts = <0 89 4>;
				xlnx,datawidth = <0x20>;
				xlnx,device-id = <0x0>;
			};
		};
		axi_gpio_dma_rstn: gpio@a0010000 {
			#gpio-cells = <3>;
			clock-names = "s_axi_aclk";
			clocks = <&zynqmp_clk 71>;
			compatible = "xlnx,axi-gpio-2.0", "xlnx,xps-gpio-1.00.a";
			gpio-controller ;
			reg = <0x0 0xa0010000 0x0 0x80>;
			xlnx,all-inputs = <0x0>;
			xlnx,all-inputs-2 = <0x0>;
			xlnx,all-outputs = <0x1>;
			xlnx,all-outputs-2 = <0x0>;
			xlnx,dout-default = <0x00000000>;
			xlnx,dout-default-2 = <0x00000000>;
			xlnx,gpio-width = <0x1>;
			xlnx,gpio2-width = <0x20>;
			xlnx,interrupt-present = <0x0>;
			xlnx,is-dual = <0x0>;
			xlnx,tri-default = <0xFFFFFFFF>;
			xlnx,tri-default-2 = <0xFFFFFFFF>;
		};
		axi_gpio_dma_start: gpio@a0020000 {
			#gpio-cells = <3>;
			clock-names = "s_axi_aclk";
			clocks = <&zynqmp_clk 71>;
			compatible = "xlnx,axi-gpio-2.0", "xlnx,xps-gpio-1.00.a";
			gpio-controller ;
			reg = <0x0 0xa0020000 0x0 0x80>;
			xlnx,all-inputs = <0x0>;
			xlnx,all-inputs-2 = <0x0>;
			xlnx,all-outputs = <0x1>;
			xlnx,all-outputs-2 = <0x0>;
			xlnx,dout-default = <0x00000000>;
			xlnx,dout-default-2 = <0x00000000>;
			xlnx,gpio-width = <0x1>;
			xlnx,gpio2-width = <0x20>;
			xlnx,interrupt-present = <0x0>;
			xlnx,is-dual = <0x0>;
			xlnx,tri-default = <0xFFFFFFFF>;
			xlnx,tri-default-2 = <0xFFFFFFFF>;
		};
	};`

HongHanPT, 2024-03-10:
Hey @dantepayne ，I meet the same problem，Have you solve it？

dantepayne, 2024-04-23:
> Hey @dantepayne ，I meet the same problem，Have you solve it？
I have solve it,it is wired because the number in "dmas" is not the number of devicetree node for axi_dma_0, you can find in xilinx_dma.c -> mm2s:0 s2mm:1,if you use rx channel,it should be 1 like this :

&amba_pl {
axidma_chrdev: axidma_chrdev@0 {
compatible = "xlnx,axidma-chrdev";
dmas = <&axi_dma_0 0>;
dma-names = "rx_channel";
};
}
&axi_dma_0 {
dma-channel@a0000030 {
xlnx,device-id = <0x0>;
};
};

Hidinpro, 2025-03-26:
@dantepayne Hello, have you tried single-channel data transfer from PL to PS? I keep getting the following errors:

axidma: axidma_dma.c: axidma_start_transfer: 317: DMA receive transaction timed out.
  
[   92.233475] xilinx-vdma 80000000.dma: Cannot stop channel 000000001473675a: 0  
[  277.228516] audit: type=1701 audit(1637342640.299:3): auid=4294967295 uid=0 gid=0 ses=4294967295 pid=2914 comm="axidma_test" exe="/home/petalinux/axidma_test" sig=6 res=1