bperez77/xilinx_axidma #141: 一个小bug及解决办法，应用层程序多次调用问题

aabb-del, 2022-01-05:
libaxidma.c中初始化时需要把设备的各个通道长度置为0，不然会影响下一次初始化

xushengrui02, 2022-01-21:
方便提供一下完整的系统的设备树嘛，我加载完驱动之后没有相应设备节点

aabb-del, 2022-01-21:
设备树节点和首页介绍给的例子是一样的

radsysyk, 2022-11-01:
> libaxidma.c中初始化时需要把设备的各个通道长度置为0，不然会影响下一次初始化

可以具体说一下在哪里置为0吗？

wangzhenhua92429, 2022-11-01:
你发给我的邮件已收到！                                ——王振华

aabb-del, 2022-11-01:
在应用层初始化函数里

xiayanqian, 2022-11-11:
你好！ 可以贴一下代码吗？我实在不知道该怎么设置哪个变量，谢谢！

aabb-del, 2022-11-11:
/* Initializes the AXI DMA device, returning a new handle to the 
  * axidma_device. */ 
 struct axidma_dev *axidma_init() 
 { 
     assert(!axidma_dev.initialized); 
  
     // Open the AXI DMA device 
     axidma_dev.fd = open(AXIDMA_DEV_PATH, O_RDWR|O_EXCL); 
     if (axidma_dev.fd < 0) { 
         perror("Error opening AXI DMA device"); 
         fprintf(stderr, "Expected the AXI DMA device at the path `%s`\n", 
                 AXIDMA_DEV_PATH); 
         return NULL;

大概在这个地方加，在同一个程序里重复运行的话，这个全局变量会影响下次的结果，需要把它memset为0，或者把结构体里面的通道数目置0。话说回来，你遇到的是这个问题吗？

xiayanqian, 2022-11-12:
应该是这个问题。我的情况是，ctrl-c打断程序，再次运行程序，此时程序无法接收到外部传输给dma的数据，再打断程序，将xilinx-axidma.ko rmmod +insmod后，运行程序，就可以接收了。应该就是多次初始化的原因。

otwaybot, 2023-04-18:
我遇到相同的問題  麻煩問下具體怎麽解決的

xiayanqian, 2023-04-20:
> 我遇到相同的問題 麻煩問下具體怎麽解決的
`struct axidma_dev *axidma_init()
{
    memset(&axidma_dev, 0, sizeof(struct axidma_dev)); //add here`

otwaybot, 2023-04-20:
感谢答复 我其实遇到另外一个问题，我start_transfer timeout 改成interrupt  这样read就可以变成阻塞的  每次read过后我也是通过devmem的方式去读s2mm_length 去获取实际数据长度 但在这个过程中我发现了一个问题 只要我程序在ctrl +c 结束  映射到驱动中的表现也就read被中断 ，然后我再次运行程序 我的 devmem的方式去读s2mm_length  的值就会不正确   我觉得不正确的原因可能是 s2mm_length 寄存器更新的次序为 想要读的长度+实际数据长度+想要读的长度+实际数据长度。。。依次循环 但是由于我两次read 使寄存器更新次序变成了  想要读的长度+想要读的长度+实际数据长度+想要读的长度+实际数据长度。。依次循环，所以导致在程序在程序再次运行后他把我想要度的长度，当成实际的长度，而实际的长度当成了想要读的长度，不知到我分析的对不对   麻烦问下有遇到过类似的问题吗  有好的解决办法吗