bperez77/xilinx_axidma #147: One-way transfer from PL to PS

eric950722, 2023-04-25:
Hi,

I'm trying to transfer data from PL to PS using the AXI DMA IP (with SG) using this xilinx_axidma driver.

The first test I did is the loopback test. For this test, I've connected the M_AXIS_MM2S port of the DMA to the slave ports of a AXI-stream FIFO. The master ports of the AXI-stream FIFO are then connected to the S_AXIS_S2MM port of the DMA.
The loopback test was successful as I'm able to transfer data from PS to PL and read back the same data from PL to PS. Here is the ILA waveform for this test:
[image]

The next test I did is to just transfer data from PL to PS as this is what I needed in my design. For this, I only have the connection between the master ports of the AXI-stream FIFO and the S_AXIS_S2MM port of the DMA. The data generation in the PL is controlled from the PS and the generated data in PL is then written to the FIFO through the slave ports of the AXI-stream FIFO. However, I observed the problem where the s_axis_s2mm_tready line remaining low after the first 4 transfers (see ILA waveform below). For this test, I'm using the `axidma_oneway_transfer()` function call instead of the `axidma_twoway_transfer()` function call.
[image]
As a result, I couldn't get any subsequent transfers and the DMA transfer eventually timeout with the following error:
`Failed to perform the AXI DMA transfer: Timer expired`
I also got the following kernel error message:
`axidma: axidma_dma.c: axidma_start_transfer: 301: DMA receive transaction timed out.`

Has anyone encounter this issue before?

If anyone had success with one-way transfer from PL to PS using this driver, it'll be appreciated if you can share how you did it.

Thanks in advance.

hamidrmt, 2023-11-21:
I have same issue, in pl before dma initialization give data to dma and then it exit synchronisation (according to tlast)

kdurant, 2024-08-21:
@eric950722  I have same issue. Have you solved the problem yet?

sudi233, 2025-09-24:
@kdurant Have you solved the problem yet