bperez77/xilinx_axidma #148: axidma: axidma_chrdev.c: axidma_mmap: 294: Unable to allocate contiguous DMA memory region of size 8294400.

L123ee, 2023-07-15:
When i test. " axidma-benchmark" application, print this error:
"""
AXI DMA Benchmark Parameters:
        Transmit Buffer Size: 7.91 Mb
        Receive Buffer Size: 7.91 Mb
vma->vm_start = 0xb6659000
dma_alloc->user_addr = b6659000
dma_alloc->dma_addr=0x72612d78
vma->vm_page_prot = 0x71f

        Number of DMA Transfers: 1000 transfers

axidma axidma: coherent DMA mask is unset
 dma_alloc->size = 0x007e9000
axidma: axidma_chrdev.c: axidma_mmap: 294: Unable to allocate contiguous DMA memory region of size 8294400.
axidma: axidma_chrdev.c: axidma_mmap: 296: Please make sure that you specified cma=<size> on the kernel command line, and the size is large enough.
Unable to allocate transmit buffer from the AXI DMA device.: Cannot allocate memory
"""
```
root@axidma_7045:/# cat /proc/meminfo
MemTotal:        1030296 kB
MemFree:          998232 kB
MemAvailable:     988188 kB
Buffers:               0 kB
Cached:            15516 kB
SwapCached:            0 kB
Active:             9404 kB
Inactive:           8392 kB
Active(anon):       9404 kB
Inactive(anon):     8392 kB
Active(file):          0 kB
Inactive(file):        0 kB
Unevictable:           0 kB
Mlocked:               0 kB
HighTotal:        262144 kB
HighFree:         242904 kB
LowTotal:         768152 kB
LowFree:          755328 kB
SwapTotal:             0 kB
SwapFree:              0 kB
Dirty:                 0 kB
Writeback:             0 kB
AnonPages:          2324 kB
Mapped:             4768 kB
Shmem:             15516 kB
Slab:               9860 kB
SReclaimable:       6200 kB
SUnreclaim:         3660 kB
KernelStack:         384 kB
PageTables:          192 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:      515148 kB
Committed_AS:      30796 kB
VmallocTotal:     245760 kB
VmallocUsed:           0 kB
VmallocChunk:          0 kB
CmaTotal:          32768 kB
CmaFree:           32436 kB
root@axidma_7045:/# dmesg | grep cma
cma: early_cma(32M)
cma: dma_contiguous_reserve(limit ffffffff)
cma: dma_contiguous_reserve: reserving 32 MiB for global area
cma: cma_declare_contiguous(size 0x02000000, base 0x00000000, limit 0xffffffff alignment 0x00000000)
cma: Reserved 32 MiB at 0x3e000000
Kernel command line: console=ttyPS0,115200 earlyprintk cma=32M
Memory: 990548K/1048576K available (6144K kernel code, 238K rwdata, 1568K rodata, 1024K init, 154K bss, 25260K reserved, 32768K cma-reserved, 229376K highmem)
cma: cma_alloc(cma c0a4d0e0, count 64, align 6)
cma: cma_alloc(): returned effb8000
cma: cma_alloc(cma c0a4d0e0, count 1, align 0)
cma: cma_alloc(): returned effb8800
cma: cma_alloc(cma c0a4d0e0, count 8, align 3)
cma: cma_alloc(): returned effb8900
cma: cma_alloc(cma c0a4d0e0, count 1, align 0)
cma: cma_alloc(): returned effb8820
cma: cma_alloc(cma c0a4d0e0, count 8, align 3)
cma: cma_alloc(): returned effb8a00
cma: cma_alloc(cma c0a4d0e0, count 1, align 0)
cma: cma_alloc(): returned effb8840
axidma: axidma_chrdev.c: axidma_mmap: 296: Please make sure that you specified cma=<size> on the kernel command line, and the size is large enough.
axidma: axidma_chrdev.c: axidma_mmap: 296: Please make sure that you specified cma=<size> on the kernel command line, and the size is large enough.
axidma: axidma_chrdev.c: axidma_mmap: 296: Please make sure that you specified cma=<size> on the kernel command line, and the size is large enough.
axidma: axidma_chrdev.c: axidma_mmap: 296: Please make sure that you specified cma=<size> on the kernel command line, and the size is large enough.
axidma: axidma_chrdev.c: axidma_mmap: 296: Please make sure that you specified cma=<size> on the kernel command line, and the size is large enough.
```

32M of CMA memory has been configured in the kernel。

what may be the above problem and how to solve it?

hanie-gh, 2023-11-01:
You can increase the CMA size in the Linux kernel with the following bootargs in your device tree:
`chosen {
        setenv bootargs " earlycon console=ttyPS1,115200 clk_ignore_unused root=/dev/ram0 rw init_fatal_sh=1 cma=1200M"
    };`

wangzhenhua92429, 2023-11-01:
你发给我的邮件已收到！                                ——王振华

Yuxin-Yu, 2024-07-11:
hi @L123ee .Have you solved your problem? My CMA capacity is 256MB and I also get the same error
```
root@debian:/proc# dmesg | grep cma
[    0.000000] cma: Reserved 256 MiB at 0x00000000f0000000
[    0.000000] Memory: 3774496K/4192256K available (7046K kernel code, 4956K rwdata, 4096K rodata, 2158K init, 316K bss, 155616K reserved, 262144K cma-reserved)
[ 1545.942406] axidma: axidma_chrdev.c: axidma_mmap: 283: Please make sure that you specified cma=<size> on the kernel command line, and the size is large enough.
[ 1549.318456] axidma: axidma_chrdev.c: axidma_mmap: 283: Please make sure that you specified cma=<size> on the kernel command line, and the size is large enough.
```