bperez77/xilinx_axidma #149: axidma_oneway_transfer() transfer issue

mahhww, 2023-08-03:
I used the method axidma_oneway_transfer() to individually write a 1024 bytes of data and then read a 1024 bytes of data, and the result is what I expected.
But when I write a 1024 bytes data with this method and read the 512 bytes data twice, I can read the data out the first time but the DMAIntErr of the S2MM_DMASR Register will be set to 1 and I can't read the data the second time and I can't send the data again.
When I write two 1024 bytes of data with this method and read 2048 bytes of data once more, I can see through the Direct Register Mode Register that S2MM_LENGTH is 1024, not 2048. here I can be sure that I am handing it the 2048 parameter and can catch this through vivado instruction signal, but when I read S2MM_LENGTH, it is 1024.
This problem has been bothering me for a month now, I don't think the driver should have this problem. Thank you very much!

wangzhenhua92429, 2023-08-03:
你发给我的邮件已收到！                                ——王振华

nandy222, 2024-03-06:
I have the same problem. Have you solved it?

mahhww, 2024-03-12:
解决了，我的问题是设置的strm2mm数据位宽有问题。

mahhww, 2024-03-12:
> I have the same problem. Have you solved it?

解决了，我的问题是设置的strm2mm数据位宽有问题。