bperez77/xilinx_axidma #150: Petalinux wont build - Error at axidma_chrdev node in system-user.dtsi

Sujjan19, 2023-09-08:
I'm using petalinux 2018.1 for a custom project, i was trying to bring up axidma as /dev/axidma/ .
I followed this issue #24 and copied all the files and also made changes to xilinx-axidma.bb file as suggested. But when i build the project after that, it fails pointing at axidma_chrdev node line in system-user.dtsi :

`ERROR: device-tree-xilinx+gitAUTOINC+682c126ef6-r0 do_compile: Function failed: do_compile (log file is located at /home/qunu/sujan/pcie_qrng/qrng_peta/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+682c126ef6-r0/temp/log.do_compile.40485)
ERROR: Logfile of failure stored in: /home/qunu/sujan/pcie_qrng/qrng_peta/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+682c126ef6-r0/temp/log.do_compile.40485
Log data follows:
| DEBUG: Executing shell function do_compile
| **Error: /home/qunu/sujan/pcie_qrng/qrng_peta/build/../components/plnx_workspace/device-tree/device-tree/pl.dtsi:75.40-41 syntax error**
| FATAL ERROR: Unable to parse input tree
| WARNING: /home/qunu/sujan/pcie_qrng/qrng_peta/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+682c126ef6-r0/temp/run.do_compile.40485:1 exit 1 from 'dtc -I dts -O dtb -R 8 -p 0x1000 -b 0 -i /home/qunu/sujan/pcie_qrng/qrng_peta/build/../components/plnx_workspace/device-tree/device-tree -i /home/qunu/sujan/pcie_qrng/qrng_peta/build/tmp/work-shared/plnx-zynqmp/kernel-source/include -i /home/qunu/sujan/pcie_qrng/qrng_peta/build/tmp/work-shared/plnx-zynqmp/kernel-source/arch/arm64/boot/dts/xilinx -i /home/qunu/sujan/pcie_qrng/qrng_peta/build/tmp/work-shared/plnx-zynqmp/kernel-source/include -i /home/qunu/sujan/pcie_qrng/qrng_peta/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+682c126ef6-r0 -o ${DTS_NAME}.dtb `basename ${DTS_FILE}`.pp'
| ERROR: Function failed: do_compile (log file is located at /home/qunu/sujan/pcie_qrng/qrng_peta/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+682c126ef6-r0/temp/log.do_compile.40485)
ERROR: Task (/home/ranjan/PCIE_QRNG/petalinux_tool/components/yocto/source/aarch64/layers/meta-xilinx/meta-xilinx-bsp/recipes-bsp/device-tree/device-tree.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3072 tasks of which 3064 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/ranjan/PCIE_QRNG/petalinux_tool/components/yocto/source/aarch64/layers/meta-xilinx/meta-xilinx-bsp/recipes-bsp/device-tree/device-tree.bb:do_compile
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
ERROR: Failed to build project`

Here is my system-user.dtsi file:

`/include/ "system-conf.dtsi"
/ {
};

axidma_chrdev: axidma_chrdev@0 {
                compatible = "xlnx,axidma-chrdev";
                dmas = <&Path1_DMA 1>;
                dma-names = "rx_channel";
        };
&Path1_DMA {
        status = "okay";
        dma-channel@80004030 {
                xlnx,device-id = <0x0>;
                //dma-coherent;
        };
};`

Note: The DMA design has only rx channel.

Please help.

wangzhenhua92429, 2023-09-08:
你发给我的邮件已收到！                                ——王振华

Sujjan19, 2023-09-13:
I found the solution 
Modify the parent node of axidma_chrdev like this:
```

/{ 
   axidma_chrdev: axidma_chrdev@0 {
                compatible = "xlnx,axidma-chrdev";
                dmas = <&Path1_DMA 1>;
                dma-names = "rx_channel";
};
```

closed 2023-09-13 by Sujjan19