bperez77/xilinx_axidma #151: segmentation fault on insmod

Sujjan19, 2023-09-08:
I am getting segmentation fault on running `insmod xilinx-axidma.ko` :

[  702.381163] xilinx_axidma: loading out-of-tree module taints kernel.
[  702.388131] Unable to handle kernel NULL pointer dereference at virtual address 00000000
[  702.396157] Mem abort info:
[  702.398910]   Exception class = DABT (current EL), IL = 32 bits
[  702.404818]   SET = 0, FnV = 0
[  702.407853]   EA = 0, S1PTW = 0
[  702.410973] Data abort info:
[  702.413843]   ISV = 0, ISS = 0x00000006
[  702.417662]   CM = 0, WnR = 0
[  702.420616] user pgtable: 4k pages, 39-bit VAs, pgd = ffffffc07b3ee000
[  702.427125] [0000000000000000] *pgd=000000007b3ab003, *pud=000000007b3ab003, *pmd=0000000000000000
[  702.436077] Internal error: Oops: 96000006 [#1] SMP
[  702.440921] Modules linked in: xilinx_axidma(O+) uio_pdrv_genirq
[  702.446914] CPU: 1 PID: 2273 Comm: insmod Tainted: G           O    4.14.0-xilinx-v2018.1 #1
[  702.455332] Hardware name: xlnx,zynqmp (DT)
[  702.459496] task: ffffffc07a41c580 task.stack: ffffff800d280000
[  702.465415] PC is at axidma_of_parse_dma_nodes+0x3e4/0x5b8 [xilinx_axidma]
[  702.472269] LR is at axidma_of_parse_dma_nodes+0x254/0x5b8 [xilinx_axidma]
[  702.479115] pc : [<ffffff8000a5ad7c>] lr : [<ffffff8000a5abec>] pstate: 60000145
[  702.486493] sp : ffffff800d2839e0
[  702.489788] x29: ffffff800d2839e0 x28: ffffff8000a5cc40 
[  702.495084] x27: ffffffc07fef4d88 x26: ffffffc07a5a2ea0 
[  702.500379] x25: ffffffc07a5a2e80 x24: 0000000000000001 
[  702.505674] x23: 0000000000000020 x22: ffffffc07b3d9200 
[  702.510969] x21: 0000000000000000 x20: 0000000000000001 
[  702.516264] x19: ffffffc07feef850 x18: 0000000000000001 
[  702.521558] x17: 00000000004a59b0 x16: ffffff800810b398 
[  702.526853] x15: ffffffffffffffff x14: 0000000000000000 
[  702.532148] x13: 0000000000000010 x12: 0101010101010101 
[  702.537443] x11: 0000000000000010 x10: 0101010101010101 
[  702.542738] x9 : 0000000000000002 x8 : 7f7f7f7f7f7f7f7f 
[  702.548033] x7 : 6d2c606c62ff726b x6 : 0000000000000000 
[  702.553327] x5 : 0000000000000001 x4 : 0000000000000000 
[  702.558622] x3 : 0000000000000069 x2 : ffffff8000a5b4e0 
[  702.563917] x1 : ffffff8000a5c772 x0 : 0000000000000000 
[  702.569213] Process insmod (pid: 2273, stack limit = 0xffffff800d280000)
[  702.575896] Call trace:
[  702.578327] Exception stack(0xffffff800d2838a0 to 0xffffff800d2839e0)
[  702.584753] 38a0: 0000000000000000 ffffff8000a5c772 ffffff8000a5b4e0 0000000000000069
[  702.592566] 38c0: 0000000000000000 0000000000000001 0000000000000000 6d2c606c62ff726b
[  702.600378] 38e0: 7f7f7f7f7f7f7f7f 0000000000000002 0101010101010101 0000000000000010
[  702.608190] 3900: 0101010101010101 0000000000000010 0000000000000000 ffffffffffffffff
[  702.616002] 3920: ffffff800810b398 00000000004a59b0 0000000000000001 ffffffc07feef850
[  702.623814] 3940: 0000000000000001 0000000000000000 ffffffc07b3d9200 0000000000000020
[  702.631627] 3960: 0000000000000001 ffffffc07a5a2e80 ffffffc07a5a2ea0 ffffffc07fef4d88
[  702.639439] 3980: ffffff8000a5cc40 ffffff800d2839e0 ffffff8000a5abec ffffff800d2839e0
[  702.647251] 39a0: ffffff8000a5ad7c 0000000060000145 ffffff800d2839e0 ffffff8000a5abec
[  702.655063] 39c0: 0000008000000000 0000000000000001 ffffff800d2839e0 ffffff8000a5ad7c
[  702.662886] [<ffffff8000a5ad7c>] axidma_of_parse_dma_nodes+0x3e4/0x5b8 [xilinx_axidma]
[  702.670784] [<ffffff8000a5a53c>] axidma_dma_init+0xa4/0x280 [xilinx_axidma]
[  702.677726] [<ffffff8000a58078>] axidma_probe+0x40/0xf0 [xilinx_axidma]
[  702.684318] [<ffffff8008583410>] platform_drv_probe+0x58/0xb8
[  702.690043] [<ffffff80085817fc>] driver_probe_device+0x22c/0x2d8
[  702.696032] [<ffffff8008581964>] __driver_attach+0xbc/0xc0
[  702.701499] [<ffffff800857f93c>] bus_for_each_dev+0x4c/0x98
[  702.707054] [<ffffff80085810f8>] driver_attach+0x20/0x28
[  702.712350] [<ffffff8008580c48>] bus_add_driver+0x1b8/0x228
[  702.717905] [<ffffff8008582310>] driver_register+0x60/0xf8
[  702.723374] [<ffffff8008583360>] __platform_driver_register+0x40/0x48
[  702.729807] [<ffffff8000a62018>] axidma_init+0x18/0x1000 [xilinx_axidma]
[  702.736482] [<ffffff8008083980>] do_one_initcall+0x38/0x128
[  702.742037] [<ffffff800810bcd0>] do_init_module+0x58/0x1b0
[  702.747504] [<ffffff800810ac44>] load_module+0x19fc/0x2150
[  702.752972] [<ffffff800810b490>] SyS_init_module+0xf8/0x178
[  702.758526] Exception stack(0xffffff800d283ec0 to 0xffffff800d284000)
[  702.764951] 3ec0: 000000001113a2e0 000000000000aee0 000000001113a260 000000001113a280
[  702.772765] 3ee0: 0000000000000000 0000007f8d5d1580 0000007f8d6ac700 0000007f8d6ac700
[  702.780577] 3f00: 0000000000000069 000000000000b001 0101010101010101 0000000000000000
[  702.788389] 3f20: 0000000000000018 0000000000005b76 0000007f8d494d98 0000007f8d487d00
[  702.796201] 3f40: 0000007f8d54ca80 00000000004a59b0 000000000000073f 0000007f8d6ac720
[  702.804013] 3f60: 000000001113a2e0 000000001113a260 0000000000000002 0000007fd3ff33f8
[  702.811825] 3f80: 0000000000000049 0000000000000000 00000000004a5000 0000000000000000
[  702.819637] 3fa0: 0000000000000000 0000007fd3ff32c0 000000000040f368 0000007fd3ff31a0
[  702.827449] 3fc0: 0000007f8d54caa4 0000000080000000 000000001113a2e0 0000000000000069
[  702.835261] 3fe0: 0000000000000000 0000000000000000 0000000000000000 0000000000000000
[  702.843074] [<ffffff80080837b0>] el0_svc_naked+0x24/0x28
[  702.848367] Code: 58000ee2 2a1803e5 58000961 52800d23 (f94002a4) 
[  702.854442] ---[ end trace 5233cb4a2c13e94a ]---
Segmentation fault

Note : My dma design only has rx channel 

My system-user.dtsi:

```
/include/ "system-conf.dtsi"
/ {
};

/{
axidma_chrdev: axidma_chrdev@0 {
                compatible = "xlnx,axidma-chrdev";
                dmas = <&Path1_DMA 1>;
                dma-names = "rx_channel";
        };
};
&Path1_DMA {
        status = "okay";
        dma-channel@80004030 {
                xlnx,device-id = <0x0>;
                //dma-coherent;
        };
};
```

My device tree :

```
/dts-v1/;
{
 dma@80004000 {
                        #dma-cells = <0x1>;
                        clock-names = "s_axi_lite_aclk", "m_axi_sg_aclk", "m_axi_s2mm_aclk";
                        clocks = "s_axi_lite_aclk", "m_axi_sg_aclk", "m_axi_s2mm_aclk";
                        compatible = "xlnx,axi-dma-1.00.a";
                        reg = <0x0 0x80004000 0x0 0x1000>;
                        xlnx,addrwidth = <0x20>;
                        status = "okay";
                        linux,phandle = <0x33>;
                        phandle = <0x33>;

                        dma-channel@80004030 {
                                compatible = "xlnx,axi-dma-s2mm-channel";
                                dma-channels = <0x1>;
                                xlnx,datawidth = <0x20>;
                                xlnx,device-id = <0x0>;
                        };
                };

   axidma_chrdev@0 {
                compatible = "xlnx,axidma-chrdev";
                dmas = <0x33 0x1>;
                dma-names = "rx_channel";
        };
};
```
Please help.

hamidrmt, 2023-11-21:
I have same issue, You should enable two dma channels and give right device-id in dts file and dmas property in chrdev should start with 0:
dmas = <&axi_dma_0 0 &axi_dma_0 1> correct
dmas = < &axi_dma_0 1> not correct

closed 2024-01-05 by Sujjan19

biyantingfeng1, 2024-04-22:
> I have same issue, You should enable two dma channels and give right device-id to dma channels in dts file and dmas property in chrdev should start with 0: dmas = <&axi_dma_0 0 &axi_dma_0 1> correct dmas = < &axi_dma_0 1> not correct

Hi, hamidrmt, very thanks for the way you giving to solve this issue. I have a try but it didn't work well(also getting segmentation fault on running .ko). In PL, I only enable write channel, even so, should I enable two dma channels in system-user.dtsi or just one I needed?