bperez77/xilinx_axidma #153: Unable to handle kernel NULL pointer dereference at virtual address 00000000

Cuty-pink, 2023-11-21:
According to the operations in the readme, I generated a .ko file and an App file. However, when I tried to load the driver on the development board, I encountered an error. Can you help me figure out what the problem is?
[image]

hamidrmt, 2023-11-21:
Ihave same issue, You should enable two dma channels and give right device-id in dts file

Cuty-pink, 2023-11-22:
You are right! Thanks for your suggest!

closed 2023-11-22 by Cuty-pink

WalkingSeason, 2024-04-15:
I have another solution, for builds that axidma doesn't have channel 0. I commented out lines 98-100 of /driver/axidma_of.c :
```c
    // if (channel == 1) {
        // dma_chan_node = of_get_next_child(dma_node, dma_chan_node);
    // }
```

then it works well.

[image]

Here is my device-tree:

```device-tree
&amba_pl{

    /* Comaptible point for axidma for bmperez axidma drivers */
    axidma_chrdev: axidma_chrdev@0 {
        compatible = "xlnx,axidma-chrdev";
        dmas = <&axi_dma_2 0 &axi_dma_3 0 &axi_dma_0 1 &axi_dma_1 1>;
        dma-names = "dma2", "dma3", "dma0", "dma1";
    };
}
```

wangzhenhua92429, 2024-04-15:
你发给我的邮件已收到！                                ——王振华