bperez77/xilinx_axidma #154: Can axidma_video_transfer work? Kernel stack is corrupted in: axidma_prep_transfer+0x2b8/0x308

La-Nausee, 2024-04-17:
Hi,

I am testing VDMA with your driver code on petalinux-v2021.2, the Linux version of which is 5.10.0. When calling axidma_video_transfer, the kernel is always corrupted, and the error message shows

**Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: axidma_prep_trasfer + 0x2b8/0x308.**

Via debugging with printk, the corruption is actually happened in the function of dmaengine_prep_interleaved_dma. I don't know how to track the further reason for the corruption, does anybody know the solution?

By the way, DMA works very well.

Yuxin-Yu, 2024-07-11:
Hi @La-Nausee ,Have you solved this error,I met the same problem.

La-Nausee, 2024-07-11:
> Hi @La-Nausee @bperez77 ,Have you solved this error,I met the same problem.

Not yet. It is due to the Linux version

wangzhenhua92429, 2025-09-18:
你发给我的邮件已收到！                                ——王振华

jennib3, 2025-11-22:
> I found the cause of the error—it resulted from incorrect use of the dma_interleaved_template structure in the axidma_prep_transfer function.

@wangyize1991  Any chance of sharing the fix?

Thanks!